MatterHackers/MatterControl #3617: Bug? When Printing Extruder Moves To Ghost Wipe Tower

smorrison047, 2018-08-11:
While printing a model (slice settings everywhere show that the Wipe Tower option is "0" for disabled), once the layer is finished, the printer goes to the left front corner of the bed as if it is performing a wipe operation of the nozzle.  I checked, re-checked, and checked again to make sure that this _**feature is not active**_ in any way or anywhere within the software.  

I hope I am describing this intelligibly.  

Thank you.

smorrison047, 2018-08-12:
Ok, will do, thanks Lars

smorrison047, 2018-08-12:
[code print_log ghost wipe.txt](https://github.com/MatterHackers/MatterControl/files/2280648/code.print_log.ghost.wipe.txt)

[image]

[Zip of settings etc.zip](https://github.com/MatterHackers/MatterControl/files/2280649/Zip.of.settings.etc.zip)

smorrison047, 2018-08-13:
Updated the title to be a more accurate description of the issue.

After running several prints and actually watching those prints in their entirety the following was observed:
- After the completion of each layer, the printer "Homes" the X and Y Axis then proceeds to print the next layer.

Please accept my apologies for any inconsistencies or bad explanations.  I will also start a new topic so that this can be properly addressed.

Thank you.

smorrison047, 2018-08-27:
**Also posted under corrected topic**

After going through setting after setting, option after option one by one and restarting the program, I finally succeeded in stopping the process of 'homing X and Y after each printed layer completed'.  To stop this behavior, I turned off "Home Z Max" under Printer - Features - Home Z Max button to off/grayed, then restarted the program.  Restarting the program was what actually stopped the action after deactivating it.  Having just deactivated the option (and not restarting the program) then printing again did not change the X and Y Homing after each layer completed.

It is and was my understanding that this was a feature similar to that of "Z Safe Homing".  It was also my understanding that the "Home Z Max" would be a feature for when the print was completed.

So either way, I am not sure if my understanding was incorrect OR if this is what "Home Z Max" was suppose to be.

Regardless, I have stopped it and leave the rest of it in your capable hands.

If someone can let me know about this, I would greatly appreciate it.  And as always, if this was user error - I own it!

Thanks guys.

smorrison047, 2018-08-27:
After MC updated today, guess what... Its back, same problem happening.  Please HELP!!!

jlewin, 2018-08-29:
Regressed in subsequent fixes, should be fixed in the next release

closed 2018-08-31 by larsbrubaker