MatterHackers/MatterControl #3620: Correction to #3617 Bug? **New Description** - Printer Homes X and Y Axis After Each Layer Completes

smorrison047, 2018-08-13:
During any print - after the completion of a layer, the printer "Homes" the X and Y Axis then proceeds to print the next layer.

Initially I believed this to be performing a "Wipe" of the extruder and referred to it as a "Ghost Wipe"  due to this being a dual extruder machine.  

Please accept my apologies for any inconsistencies or bad explanations on issue #3617 .  I have also added a new note to that topic and modified the title as well.

Thank you.
[Zip of settings etc.zip](https://github.com/MatterHackers/MatterControl/files/2283429/Zip.of.settings.etc.zip)
[print_log.txt](https://github.com/MatterHackers/MatterControl/files/2283438/print_log.txt)
[Zip of settings etc.zip](https://github.com/MatterHackers/MatterControl/files/2283439/Zip.of.settings.etc.zip)
[image]
[code print_log ghost wipe.txt](https://github.com/MatterHackers/MatterControl/files/2283440/code.print_log.ghost.wipe.txt)
[From an earlier print attempt.txt](https://github.com/MatterHackers/MatterControl/files/2283441/From.an.earlier.print.attempt.txt)

smorrison047, 2018-08-13:
[image]

smorrison047, 2018-08-19:
UPDATE 08/19/2018:

After numerous prints and going through every single item I am able inside of MC V2, my firmware, config.h and config.adv.h, I am unable to correct this issue.  The X and Y Axis are definitely "homing" after the completion of each layer that prints.

Result: This is causing very sloppy prints and making a stringing mess.  I did get the stringing down to a minimum with retraction which now leaves very fine hair like strings.  The prints are still messy none-the-less.

I know and appreciate how busy you guys are:
I don't know if anyone else is having this problem, or if you can offer any suggestions on settings I can check, anything to try would be great as this is so annoying and i really hope it is something simple.  If this turns out to be user error, I am going to own that shit like a champ!

smorrison047, 2018-08-27:
After going through setting after setting, option after option one by one and restarting the program, I finally succeeded in stopping the process of 'homing X and Y after each printed layer completed'.  To stop this behavior, I turned off "Home Z Max" under Printer - Features - Home Z Max button to off/grayed, then restarted the program.  Restarting the program was what actually stopped the action after deactivating it.  Having just deactivated the option (and not restarting the program) then printing again did not change the X and Y Homing after each layer completed.

It is and was my understanding that this was a feature similar to that of "Z Safe Homing".  It was also my understanding that the "Home Z Max" would be a feature for when the print was completed.

So either way, I am not sure if my understanding was incorrect OR if this is what "Home Z Max" was suppose to be.

Regardless, I have stopped it and leave the rest of it in your capable hands.

If someone can let me know about this, I would greatly appreciate it.  And as always, if this was user error - I own it!

Thanks guys.

smorrison047, 2018-08-27:
After MC updated today, guess what... Its back, same problem happening.  Please HELP!!!

jlewin, 2018-09-04:
We had some regressions in MatterSlice that were causing unexpected moves to the origin at each layer transition. At this time we believe these problems have been resolved. Are you still seeing the problem behavior?

larsbrubaker, 2018-09-23:
Closing as fixed

closed 2018-09-23 by larsbrubaker