MatterHackers/MatterControl #3624: unhandled exception

techbill56, 2018-08-13:
Good afternoon,

This screen always pops up when I first open Mattercontrol 2.0:

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.Net.WebException: The request was aborted: Could not create SSL/TLS secure channel.
   at System.Net.HttpWebRequest.EndGetResponse(IAsyncResult asyncResult)
   at System.Net.Http.HttpClientHandler.GetResponseCallback(IAsyncResult ar)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2650.0 built by: NET471REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
MatterControl
    Assembly Version: 2.0.0.9825
    Win32 Version: 2.0.0.9825
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/MatterControl.exe
----------------------------------------
Agg
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/Agg.DLL
----------------------------------------
netstandard
    Assembly Version: 2.0.0.0
    Win32 Version: 4.7.2563.0
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/netstandard/v4.0_2.0.0.0__cc7b13ffcd2ddd51/netstandard.dll
----------------------------------------
Mindscape.Raygun4Net4
    Assembly Version: 5.5.2.0
    Win32 Version: 5.5.2.0
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/Mindscape.Raygun4Net4.DLL
----------------------------------------
Mindscape.Raygun4Net
    Assembly Version: 5.5.2.0
    Win32 Version: 5.5.2.0
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/Mindscape.Raygun4Net.DLL
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2634.0 built by: NET471REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2634.0 built by: NET471REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2634.0 built by: NET471REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Gui
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/Gui.DLL
----------------------------------------
VectorMath
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/VectorMath.DLL
----------------------------------------
Newtonsoft.Json
    Assembly Version: 9.0.0.0
[... 339 lines cut ...]
    Assembly Version: 2.9.0.0
    Win32 Version: 2.9.0.15
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/Zeroconf.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

I am guessing this is just something on my end, but after speaking with Taylor, I decided to add this for you to look at.  Thanks!

Bill

larsbrubaker, 2018-08-14:
Looks like SSL might not have valid root certificates. Are your running on Linux? Can you provide more info about your system?

Thanks.

techbill56, 2018-08-14:
Good afternoon,

I am running Windows 7 professional.  It is just a generic Dell.  Thanks!

Bill

podmore71, 2018-08-25:
I'm getting the same error. Had it since I first downloaded ver 2 and it still comes up when I open Mattercontrol even after updating. Also running a generic Dell laptop and Windows 7 Pro 64 bit

podmore71, 2018-08-25:
[image]

PlastiBots, 2018-09-22:
Ditto.  Just downloaded v2.0 and same issue. Windows 7 Pro.

podmore71, 2018-09-22:
Glad its just not me then. I thought it was cause I'm running it on a crappy old Dell laptop

Regnad517, 2019-01-05:
Any further progress on this issue?  I can use 1.7 but really want to use 2 and I am having the exact same issue.  Windows 7 32bit here.

techbill56, 2019-01-05:
I have been running version 2.0 even with that issue.  After Lars recommended the Alpha build on Twitter, I switched to that build and it has been running pretty well.  When that error message pops up, I just clicked the x and went on with my business.  You will really like MatterControl 2.0.

Regnad517, 2019-01-05:
Unfortunately for me, when I click either Continue or the X, the program just completely freezes.

Regnad517, 2019-01-06:
So I have an update at least on my end.  I decided to try an entirely new computer - 64bit this time and v2 worked great - for a day.  Then I added a G29 command to my start Gcode and exception error returned.  When I removed the modified Gcode, the error remains and again, V2 is unusable,  Going to mess around some more.

unlimitedbacon, 2019-01-07:
@Regnad517, if you find that this issue is related to a particular setting or configuration, them please post your printer profile for us.

Regnad517, 2019-01-08:
Well I am now completely at a loss.  Printed 3 things yesterday, no issues.  Got home today, changed nothing, and when it gets to loading gcode - crashes again and now even restore point doesnt resolve.

I have a Monoprice Maker Select V2 using all standard defaults for now.

Basically - fresh install on 64 bit Windows 7 machine ran great for 1 day.  Everytime I get to loading Gcode while slicing, the exception error listed above hits me and I am completely frozen from that point on.

Regnad517, 2019-01-08:
I just realized I am getting a different exception error than this thread is about,  will try to find my error or open a new thread.  Apologies