MatterHackers/MatterControl #3628: Crashing

rebelskyj, 2018-08-15:
Anytime I try to send a file to a Pulse printer from matter controller the app crashes.

jlewin, 2018-08-15:
- Can you describe the steps you take to produce the crash? 
- Does it happen when you print only a single cube and nothing else?

[image]

jlewin, 2018-08-17:
Are you running 2.0? We recommend that all our pulse users are on 2.0 beta. You can get it [here](https://www.matterhackers.com/store/l/mattercontrol/sk/MKZGTDW6).

If you are already on 2.0, can you attach your settings (export them) to this issue in a zip and we'll try to get to the bottom of it.

rebelskyj, 2018-08-17:
Is the beta not available for mac?

larsbrubaker, 2018-08-22:
@rebelskyj sorry for the confusion. We are currently working on a 2.0 version of MatterControl for the Mac but it is not available at this time. Do you by chance have access to a PC so that you can be on our latest version?. 

We expect to have a beta for the mac out in a few weeks.

closed 2018-08-31 by larsbrubaker