MatterHackers/MatterControl #3634: Unhandled Exception has occurred in your application

smorrison047, 2018-08-17:
When un-grouping 2 parts on the print bed, the following happens (see pic)

[image]

Windows 10 Pro 64 bit system custom built by me
Details of the error are were copied as below:

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
   at System.ThrowHelper.ThrowArgumentOutOfRangeException(ExceptionArgument argument, ExceptionResource resource)
   at MIConvexHull.ConvexHullAlgorithm.CreateInitialSimplex()
   at MIConvexHull.ConvexHullAlgorithm.GetConvexHull()
   at MIConvexHull.ConvexHullAlgorithm.GetConvexHull[TVertex,TFace](IList`1 data, Double PlaneDistanceTolerance)
   at MatterHackers.PolygonMesh.MeshConvexHull.CreateHullMesh(Mesh mesh, VertexCollecton sourceVertices)
   at MatterHackers.PolygonMesh.MeshConvexHull.GetConvexHull(Mesh mesh, Boolean generateAsync)
   at MatterHackers.PolygonMesh.TransformedAabbCache.GetAxisAlignedBoundingBox(Mesh mesh, AxisAlignedBoundingBox verticesBounds, Matrix4X4 transform)
   at MatterHackers.MeshVisualizer.MaterialRendering.InsideBuildVolume(IObject3D item, PrinterConfig printerConfig)
   at MatterHackers.MeshVisualizer.MeshViewerWidget.GetItemColor(IObject3D item)
   at MatterHackers.MeshVisualizer.MeshViewerWidget.DrawObject(IObject3D object3D, List`1 transparentMeshes, Boolean parentSelected, DrawEventArgs e)
   at MatterHackers.MeshVisualizer.MeshViewerWidget.Draw_GlTransparentContent(Object sender, DrawEventArgs e)
   at MatterHackers.MatterControl.PartPreviewWindow.InteractionLayer.RenderSource_DrawExtra(Object sender, DrawEventArgs e)
   at System.EventHandler`1.Invoke(Object sender, TEventArgs e)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.MatterControl.PartPreviewWindow.View3DWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.MatterControl.PartPreviewWindow.PrinterTabPage.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.MatterControl.RootSystemWindow.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.WinformsSystemWindow.OnPaint(PaintEventArgs paintEventArgs)
   at MatterHackers.Agg.UI.OpenGLSystemWindow.OnPaint(PaintEventArgs paintEventArgs)
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3131.0 built by: NET472REL1LAST
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
MatterControl
    Assembly Version: 2.0.0.9832
    Win32 Version: 2.0.0.9832
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/MatterControl.exe
----------------------------------------
Agg
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/Agg.DLL
----------------------------------------
netstandard
    Assembly Version: 2.0.0.0
    Win32 Version: 4.7.3056.0
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/netstandard/v4.0_2.0.0.0__cc7b13ffcd2ddd51/netstandard.dll
[... 398 lines cut ...]
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files%20(x86)/MatterControl/MatterSlice.EXE
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3056.0 built by: NET472REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

jlewin, 2018-08-17:
Is it possible to attach the models that you are having the problem with to this bug? We are able to ungroup meshes here and not see this, so it may be something with the geometry that you are using.

smorrison047, 2018-08-17:
<div>Ok, will do that in a minute</div><div><br></div>Sent from my Verizon ASUS Smartphone

smorrison047, 2018-08-17:
[Spool_Roller MOUNT PART MOD Rev 1a-3R1 re draw .zip](https://github.com/MatterHackers/MatterControl/files/2298421/Spool_Roller.MOUNT.PART.MOD.Rev.1a-3R1.re.draw.zip)

larsbrubaker, 2018-08-17:
I'm seeing this work on the latest build. Can you make sure you are up to date and let me know if it is still broken for you?

[image]

smorrison047, 2018-08-17:
it's still happening to me, the build I have installed is .9833

I tried to change the model to a similar but completely different file, same problem.  I even tried a completely different model and same.  This also happened on one attempt after duplicating the item then moving it to a different location - however I believe that was possibly caused by my system hanging up when my virus protection (trend micro extra super premium version free from my internet/cable company) quick check started, which works randomly.  I have since turned that off.  I will re-boot my machine this evening and let you know if the same thing happens.  Also, I cleared the cache and all temp files etc.

smorrison047, 2018-08-19:
UPDATE: 08/19/2018

Have not had this issue pop up again after numerous attempts, however I will keep trying to duplicate it and will update as necessary.  

Not sure what was going on, but... it seems as though something with my system may have been causing this.  If it was my system, please accept my apologies.

smorrison047, 2018-08-20:
UPDATE: 08/20/2018

Problem does in fact continue to occur in the same manner as described above.

Profkanz, 2018-09-16:
I have seen a similar error message when loading an stl or amf file directly to the bed from a folder on the C drive. It does not occur when the file is added to Local Library first.