MatterHackers/MatterControl #3655: Hotend Temperature

Hal1949, 2018-08-27:
The temperatures for the hotends are not displayed.  I'm not sure if this is an issue or they were deliberately
 left off the Controls Panel.  I'm running MatterHackers:MatterControl 2.0.0.9857  32-bit Alpha version. My OS is Windows 7 Ultimate SP1

larsbrubaker, 2018-08-27:
You should find the hotend and bed temps in the upper right corner of the connected printer.

[image]

You can also open these to control aspects of the hotend.

[image]

Is this what you are looking for?

Hal1949, 2018-08-27:
Yes it is exactly what I’m looking for .. thanks for such a speedy response.  Weird, I didn’t even see them there …  Thanks

 

 <https://user-images.githubusercontent.com/1158332/44672533-98218400-a9dd-11e8-807a-87359e9e6936.png> image

 <https://user-images.githubusercontent.com/1158332/44672566-abccea80-a9dd-11e8-9382-ae85884d231c.png> image

closed 2018-09-01 by larsbrubaker