MatterHackers/MatterControl #3661: Feature Recommendation: Add measure tool which can be used on different areas of the model and/or parts

smorrison047, 2018-08-28:
The ability to measure parts and spaces etc. would be significantly helpful when designing a part or placing an object onto a part.  As an example, if placing an item onto another item and you want it in a specific place, it would be nice to be able to measure from Point A to Point B.  Or, if importing a part and you want to measure the diameter of a hole to verify or otherwise verify part of an item. 

Thank you!

larsbrubaker, 2020-11-11:
There is now a rudimentary measuring tool. This will improve in future versions of MatterControl.

[image]

closed 2020-11-11 by larsbrubaker

smorrison047, 2020-11-11:
Lars,

Thank you!!!  This is an incredible enhancement to an already great program.  You guys are working your $%ses off there and providing a fantastic opportunity to people for free.  You guys there at MatterHackers deserve major recognition!  As for me, I am sincerely grateful for the opportunity you guys have provided me as a Beta Tester and I take that very seriously.  I am eager to start using the new feature and will continue to provide feedback as that Beta Tester now and in the future.  Again, I am sincerely grateful - THANK YOU!

Sincerely,

Scott A. Morrison

jeromesurfs, 2020-11-20:
Yes! Rudimentary but so useful and appreciated!

gregreid, 2022-04-26:
It's not clear if the dots on the end of the measuring tool are supposed to be aligned on the center of the line you want to reference, or on the tip.  Using arrows instead of dots would make this clearer.