MatterHackers/MatterControl #3670: Load Filament function does not work on Rostock MAX when hot end is cold

unlimitedbacon, 2018-08-29:
The load filament function uses the `M302` command to disable cold extrusion prevention so that the filament can be advanced while the hot end is still heating up. However, Repetier does not implement the `M302` command correctly. https://github.com/repetier/Repetier-Firmware/issues/548

This means that the Rostock will not advance the filament while the hot end is cold and the filament does not get loaded correctly.

larsbrubaker, 2018-10-30:
I'm going to change the initial M302 to be S1 rather than S0. This will work on Marlin and Repetier and does not require we do anything to the profile.

unlimitedbacon, 2018-10-30:
That sounds like a good idea. I think we should prefer general solutions whenever possible because otherwise we would have to maintain a hack in the profile for every Repetier printer.