MatterHackers/MatterControl #3689: how to support Chinese?

WinstonMao, 2018-09-04:
I add "Chinese"to languageDict in function"CreateLanguageDict()",but the UI does't show any Chinese characters. And the text of widget is empty.
[image]

[image]

larsbrubaker, 2018-09-05:
The main thing that needs to happen is that MC needs a Chinese true type font. 

In TypeFacePrinter.cs we load up an embedded font (LiberationSansFont.Instance) line 81, this would need to be a font that has Chinese in it. We can work out the details later, but the start is just to get a .ttf font that has the right characters in it and get some results. The font renderer will use that to find the unicode characters and we should see something on the screen. I have done a test in the past that rendered Japanese, but the code was never finished.

Hope this gives you a starting point. We plan to implement Chinese in the future, but is not on our current road map.

WinstonMao, 2018-09-05:
Thanks！
It can render Chinese when I added Chinese font.

larsbrubaker, 2018-09-07:
So, do you have this working now? Could you describe what you did to get it working? Do you have a ttf that we could distribute?

WinstonMao, 2018-09-10:
Yes,it works.
I loaded the "fireflysung.svg" which converted form "fireflysung.ttf" directly.Here is the code:
		public static TypeFace Instance
		{
			get
			{
				if (instance == null)
				{
                                     //instance = new TypeFace();
                                    //instance.ReadSVG(FontData());
                                    instance = TypeFace.LoadSVG("fireflysung.svg");
                                  }
                               return instance;
			   }
		}
I downloaded the "fireflysung.ttf" from this site: http://luc.devroye.org/fonts-51316.html
The "fireflysung.ttf" also support japanese,but not support korea.It seems Adobe's source-han series fonts support CJK language too，but I haven't test yet because of its otf fonts type.

closed 2018-09-14 by larsbrubaker

anouduoluo, 2020-02-16:
> Yes,it works.
> I loaded the "fireflysung.svg" which converted form "fireflysung.ttf" directly.Here is the code:
> public static TypeFace Instance
> {
> get
> {
> if (instance == null)
> {
> //instance = new [...]

大哥，可以详细地说一下，字体放在项目的什么位置以及这段代码写在什么地方吗？谢谢！

WinstonMao, 2020-02-23:
> 大哥，可以详细地说一下，字体放在项目的什么位置以及这段代码写在什么地方吗？谢谢！

你全局搜索不就完了，字体放哪无所谓，.LoadSVG("{你的路径}")

anouduoluo, 2020-02-24:
> 你全局搜索不就完了，字体放哪无所谓，.LoadSVG("{你的路径}")

谢谢，已经解决了！