MatterHackers/MatterControl #3731: MS .NET Error

Hal1949, 2018-09-17:
Every time I try to print after creating or modifying a model I get this error.  I'm not sure if anyone else is having this problem.  I like what you are doing with MatterControl .. so I know you can correct this.  Thanks in advance.  I'm also including a snapshot of my system information.

[image]

[image]

larsbrubaker, 2018-09-18:
I'm not sure what data you are loading but the error looks like it is an out of memory error.

MatterControl 2.0 is optimized for 64 bit and you will have a much better time with large data sets if you are running on a 64 bit operating system. The system specs you submitted show you are on a 32 bit system.

That said, we are always looking to improve our memory usage and if you can attach or link to the file you were slicing and export and add your printer profile, we will look into how to improve this problem.

Hal1949, 2018-09-19:
Unfortunately I'm not at the point anymore where I can easily go to 64-bit.  I'm retired and on limited income.  So, my 32-bit has been doing me such great justice ..  :)    

Attached you shall find a WinZip file which contains the information you requested.  I have done any changes to the GCode tab on the Printer so I didn't snapshot it.  I hope this information is helpful.

[Error Info 091918.zip](https://github.com/MatterHackers/MatterControl/files/2397527/Error.Info.091918.zip)

larsbrubaker, 2018-09-26:
Thanks. Changing the title and we will track this issue for future development.