MatterHackers/MatterControl #3762: how do you assign parts of a dual color print, to different extruders?

bman2000, 2018-09-27:
I have looked all over on version 2 and cannot seem to find any way to assign the two differnt parts of my model with one part going to extruder 1 and the other part going to extrduer 2. I have a dual nozzle printer.

Thank you

larsbrubaker, 2018-09-28:
Thanks for the question, we'll work on making this more intuitive :).

You can find it here:

[image]

bman2000, 2018-09-28:
Thank you for the information.

So I guess this means that material 1 is always assigned to the first extrduer, material 2 to the second etc.?
So If I import 2 stl files, that combine into one part, I select each part like in the video, and assign materila 1 to the first part and materila 2 to second part ( stl)?

larsbrubaker, 2018-10-07:
Yes, that should do what you want.

closed 2018-10-27 by larsbrubaker