MatterHackers/MatterControl #3782: Excessive bed leveling on Pulse

martinpirringer, 2018-10-03:
Every time you change the bed temp It forces you to do a complete bed level - makes dialing in a filament annoying. I had days were I leveled the bed more than 10 times

larsbrubaker, 2018-10-04:
Thanks for the feedback. We are planning to do this in the near future. 

So that we know to look at this when we fix the issue, adding a reference to our internal bug.
MatterHackers/MCCentral#4066
Feature request: Stored bed leveling data for different bed temperatures

larsbrubaker, 2018-10-23:
This has been fixed and will be in the next update. If the bed temp has changed more than 10 degrees, there is a warning that can be dismissed.

[image]

closed 2018-10-26 by larsbrubaker