MatterHackers/MatterControl #3784: MC 2.0 Z-offset serious problem

martinpirringer, 2018-10-03:
In MC 1.7x I could define a Z-offset for each filament which was nice as PLA likes a different z-offset than PETG etc. Then I upgraded to 2.0 as I was told the problems in 1.75 where not going to get fixed. Now it kept the z-offsets from the 1.75 version and there is no way to permanently change them. Problem is for PLA the proper z-offset in MC 1.75 was -.6 in MC 2.0 due to the different bed leveling its +0.2. So if I click on PLA and forget to then go to printer and change the z-offset from -0.6 to +0.2 then I make a nice scratch into my build plate.

larsbrubaker, 2018-10-05:
Thanks for the report.

I believe the problem that is happening here is that when we have a probe we are expecting the 'Probe Z Offset' (in printer settings) and the leveling solution to be our entire solution with no other data confounding it. 

In the case of 'Use Automatic Z Probe' we should disregard (and suppress the display of) the 'Z Offset' as the only result it can be having inside of MatterControl is to be invalidating the leveling solutions calculation.

larsbrubaker, 2018-10-05:
The thing that you could do to correct this problem, if it is related to a probe, is to set all your 'Z Offsets' to 0. We will fix this is a future update as well as look into whatever problem you are having if this is not the right diagnosis.