MatterHackers/MatterControl #3801: zhop on movements

67Exec, 2018-10-07:
I print quite a few plaques and such where part of the surface needs to be smooth and is visible, then letters and/or a design continue printing to be raised. I would love to be able to set zhop on moves that don't require retraction because it inevitably seems that the nozzle will drag at some point that's visible on the finished product. I've tried babystepping, and raising my z offset, all to no avail.
Thank you

larsbrubaker, 2018-10-08:
We are planning to add Z-Hop when traveling over top surfaces. Does that seem like it would solve your case?

67Exec, 2018-10-08:
That would depend on if MatterControl would treat the surface below the
text as a top surface. Say I used black for a plaque with bronze for the
raised design while using a single extruder with a filament change.  Would
it treat the top layer before the design as a top surface?
I included a photo as an example.

Thank you,
Adam Thomas

67Exec, 2018-10-09:
> Yes, it will do what you are hoping for.

Sweet, thank you

closed 2018-10-22 by 67Exec