MatterHackers/MatterControl #3860: Files starting with # will not slice

martinpirringer, 2018-10-20:
Even though the # sign is a valid character a file name #8nutholder will not slice you can load it into MC and display it but when you hit slice you will get a message something like "this will be a very short print it will take 0 seconds" and no layers are visible in layer mode. The only illegal characters in a windows file name are <>?*| /\and Ascii control characters 0 to 31 and some combinations by themselves like just a blank space or number of blank spaces without anything else or  a period or multiple periods by themselves. So MC should be able to slice any file with a valid Windows file name. IDK what language this is written in but it probably needs to be escaped or it is maybe interpreted as an instruction instead as part of the file name and then the open returns a 0 length string I guess. HTH

martinpirringer, 2018-10-20:
I sent an example file the MH support if that helps

larsbrubaker, 2018-10-21:
Thanks for the bug report. We'll get it fixed.

larsbrubaker, 2018-12-29:
Fixed in the latest build (in alpha now)

closed 2018-12-29 by larsbrubaker