MatterHackers/MatterControl #3861: New UI

blueeagle69, 2018-10-21:
Hi.

I hate to say this, but I much prefer the previous alpha UI layout.
This new layout is driving me insane :(

larsbrubaker, 2018-10-21:
Thanks for the feedback.
Sorry to hear that you're not liking the new layout. 

Can you please describe, in detail, what it is that you're missing?

The more detail that you can give us the better. 
We want to make sure that we include all of the capabilities that you and other users are expecting. 

We are always trying to make it as easy to use for first time and existing users, so if there is something that has become more complex please let us know that as well.

blueeagle69, 2018-10-21:
Apologies for being so vague!.It was late.

I much prefer the old v2 layout where you had previous models added to the
bed appear at the top off the screen and only needed one click to.open them
on a virtual bed.
Now you have to go through several clicks in order to get to the Plating
History. And even then they can't be added to a virtual bed with a click.
I did much prefer having the past few objects appear in a row at the top.of
the screen. They could be opened possible on a virtual bed with a simple
click of the mouse.
What's more the items that now appear down the left hand side are too small
too.

Thanks :)

larsbrubaker, 2018-10-22:
Thanks for the feedback. We will look into how to make these operations more convenient in the new UI.

DkOuvE, 2018-10-28:
While were talking about the new UI, in the Solarized Light theme, the extruder temperature font colour is the same as the background colour, thus you can't see the extruder value you enter.