MatterHackers/MatterControl #3885: Import Missing

blueeagle69, 2018-10-25:
Hi.

With the latest version 2.x.10024 where is the Import option gone?
I click on the Bed button at the top of the screen,  Cut, Copy, Paste, Print & Export are there but no Import.

larsbrubaker, 2018-10-26:
Thanks for the continued feedback. 

We are attempting to consolidate the adding of parts so that we train the right behavior. But we are always interested in improving.

We continue to think about all the concerns you and other users express and work to keep improving.

Here is a feature that I hope helps you get back to opening a bed plate easier.

[image]

blueeagle69, 2018-10-26:
> and work to keep imroving.

Hi Lars.
I did notice that new option last night.
Many thanks for the updates.

Profkanz, 2018-11-06:
I agree the new 'Open' menu item is too hidden. Really ought to be in the 'Recent'-'Print'-'Export' drop down menu.