MatterHackers/MatterControl #3890: Different Purge routine

martinpirringer, 2018-10-25:
Feature Request
===============

Problem: Currently a ball of filament is dropped in the corner. Sometimes especially with PETG it does not attach to the build plate even though the other layers due so the ball follows the hotend and the gets "run over and either messes up the first layer or permanently attaches itself on my Pules to the E3d sock. I am averaging a sock every 2 prints with PETG. Also on some other prints with any material if the print uses the whole build plate (I have some that are 247x 206 at least partially - the ball gets run over by the hot end with the above results.

Solution: Either make the purge optional and/or turn it into a line kinda like a partial skirt that lies outside the print area - A line can be "run over" or crossed by the hotend without picking it up or attaching it to the E3d sock either right away or after functioning like a "snoball" to other still hot material of the 1st layer. 

I printed 6Kg r filament so far and just started my 4th 3pack of E3d socks

unlimitedbacon, 2018-10-25:
We actually switched to a priming line for the Pulse some time ago, but it sounds like you have an older profile. If you delete your current printer profile and add a new one, it will download the new settings.

The plan is to make it so that you can enable/disable the priming line in the future. However in the mean time you can change it or shut it off by editing your [Start G-Code](http://wiki.mattercontrol.com/SETTINGS/Printer/Custom_G-Code/Start_G-Code).

jlewin, 2018-10-26:
Linking to internal issue and closing.

MatterHackers/MCCentral#4383
Add ability to turn on and off a priming line at the front of the bed

closed 2018-10-26 by jlewin