MatterHackers/MatterControl #3952: Cancel print should turn bed and extruder off

martinpirringer, 2018-11-05:
Bug Report
============
**Steps to Reproduce**
1.  Cancel a print
2. 
3. 

**Expected Behavior**

Bed and Hotend should turn off

**Actual Behavior**
They do not. Maybe the reason is that they were turned on while slicing. I am doing a lot of complicated prints that take about as long to slice as it takes the bed to warm up So I hit print and turn on the bed and nozzle while it slices to save a little time. When the print is done it shuts everything off when I cancle the print it does not

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.0.0.10028
**Operating System Version**
Win10
**Printer Make/Model**
Pulse C232

larsbrubaker, 2018-11-05:
I you cancel the print while it is slicing it does not turn off the heaters. If you cancel after the print has started it should turn them off. In our testing suit this is validated on every build and I just tested on the latest build. Is it possible this is the behavior you are seeing?

If not, I will continue to investigate. Also, any aditional informaition you can provide will be helpfull.

martinpirringer, 2018-11-05:
Addendum - It will eventually shut down - don't know after how many minutes. But had a fail that required an edit of the IPT and I left it on for testing purposes after completing the change it said "auto shutdonw in 4 min 26 sec" So IDK if that is what is intended. Usually if you have a fail you want to remove a print and that is usually easier from a build plate that cooled down a little.