MatterHackers/MatterControl #3973: Mattercontrol interferes with SD card print

martinpirringer, 2018-11-09:
Bug Report
============
**Steps to Reproduce**
1. 
2. 
3. 

**Expected Behavior**

**Actual Behavior**

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")

**Operating System Version**

**Printer Make/Model**

unlimitedbacon, 2018-11-09:
That was a very long and confusing way of describing a simple problem. Please try to stick to the bug report format. It makes it much easier for us to understand what the problem is. Here is an example.

**Steps to Reproduce**
1. Connect MatterControl to the printer.
2. Use the printer's onboard screen to start a print from the SD card.
3. Close MatterControl.

**Expected Behavior**
The print should continue uninterrupted.

**Actual Behavior**
MatterControl shuts off the heaters before closing, which causes the print to fail.

This should not be a difficult problem to solve. MatterControl could check the printer's status using the [M27 command](http://marlinfw.org/docs/gcode/M027.html). If a print is in progress then the heaters should not be turned off.

larsbrubaker, 2018-11-10:
Worked in first test case. Will look into it more.

martinpirringer, 2018-11-12:
Will send Video to support@matterhackers.com as this doesn't take videos - or I haven't figured out how to upload one

unlimitedbacon, 2018-11-15:
I can confirm that this is still happening. @larsbrubaker, did you start the print through MatterControl or from the printer itself?

closed 2018-12-08 by larsbrubaker