MatterHackers/MatterControl #3984: Slicing Bug on T100

martinpirringer, 2018-11-12:
Bug Report
============
**Steps to Reproduce**
1. Slice Mold for dual Rack (STL available if you tell me how to upload/send it)
2. Hit print when getting to write layer nn error occurs

3. 

**Expected Behavior**
Printing happening
**Actual Behavior**
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.OutOfMemoryException: Exception of type 'System.OutOfMemoryException' was thrown.
   at MatterHackers.VectorMath.WorldView.GetRayForLocalBounds(Vector2 localPosition)
   at MatterHackers.VectorMath.WorldView.GetWorldUnitsPerScreenPixelAtPosition(Vector3 worldPosition, Double maxRatio)
   at MatterHackers.RenderOpenGl.GLHelper.Render3DLineNoPrep(WorldView world, Frustum clippingFrustum, Vector3 start, Vector3 end, Color color, Double width)
   at MatterHackers.Plugins.EditorTools.ScaleCornerControl.DrawGlContent(DrawGlContentEventArgs e)
   at MatterHackers.MeshVisualizer.MeshViewerWidget.DrawInteractionVolumes(DrawEventArgs e)
   at MatterHackers.MeshVisualizer.MeshViewerWidget.Draw_GlTransparentContent(Object sender, DrawEventArgs e)
   at MatterHackers.MatterControl.PartPreviewWindow.InteractionLayer.OnDrawGlContent(DrawEventArgs e)
   at MatterHackers.MatterControl.PartPreviewWindow.InteractionLayer.RenderSource_DrawExtra(Object sender, DrawEventArgs e)
   at System.EventHandler`1.Invoke(Object sender, TEventArgs e)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.MatterControl.PartPreviewWindow.View3DWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.MatterControl.PartPreviewWindow.PrinterTabPage.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.GuiWidget.OnDraw(Graphics2D graphics2D)
   at MatterHackers.MatterControl.RootSystemWindow.OnDraw(Graphics2D graphics2D)
   at MatterHackers.Agg.UI.WinformsSystemWindow.OnPaint(PaintEventArgs paintEventArgs)
   at MatterHackers.Agg.UI.OpenGLSystemWindow.OnPaint(PaintEventArgs paintEventArgs)
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.3190.0 built by: NET472REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
MatterControl
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files/MatterControl/MatterControl.exe
----------------------------------------
MatterControl.Printing
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Program%20Files/MatterControl/MatterControl.Printing.DLL
----------------------------------------
netstandard
    Assembly Version: 2.0.0.0
    Win32 Version: 4.7.3056.0
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/netstandard/v4.0_2.0.0.0__cc7b13ffcd2ddd51/netstandard.dll
[... 461 lines cut ...]

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.0.0.10052

**Operating System Version**
Windows 10 on ASUS T100 
**Printer Make/Model**

jlewin, 2018-11-12:
Please attach the zipped `STL` files using the instructions below the edit text field
>[image]

martinpirringer, 2018-11-12:
[Mold_doubleRack_10in_V2+015.zip](https://github.com/MatterHackers/MatterControl/files/2573523/Mold_doubleRack_10in_V2%2B015.zip)
Great that zip files work. It finally sliced on my big computer and I am running GCODE of the micro SD right now. USE Pro Nylon on pulse 232 to slice with the following changes. Temp 285, Bed 80, increase extrusion multiplier by 5% whatever your machine needs, No support no brim, no raft no skirt set external layer to 1% yes one - if I could set it to 0 I would. To fit on build plate you have to rotate by -40 degrees as its 267 mm long. It is a mold for a double rack of a rack and pinion for annealing purposes printed in Nylon for heat resistance. The mold was done by a subtractive merge of 2 bodies the rack and a 2nd solid that is the outside of the mold and then split in half. It is printed upright so that the omission of the external perimeter provides the extra space to account for overprint of both the mold and the to be annealed item. It is 1.5 % oversized (scaled in inventor) to account for shrinkage when annealing the Nylon to prevent further dimensional changes in repeated heating and cooling. And I forgot set all speeds so that any speed that is above 45 is 45

martinpirringer, 2018-11-12:
Gcode failed printer stopped and made a noise the same it makes now at every sd card (not MC controlled) print at the beginning I sent a video of it to support I need a way to track the GCODE when printing from SD card - is there one?
[SD Card printing problem.zip](https://github.com/MatterHackers/MatterControl/files/2573730/SD.Card.printing.problem.zip)

martinpirringer, 2018-11-12:
and here the Gcode
[Mold10inRack+015.zip](https://github.com/MatterHackers/MatterControl/files/2573742/Mold10inRack%2B015.zip)
Now this has been created on another more powerful computer unfortunaetely I need that one for lots of other things and cannot tether the pulse to it. Hence if the T100 cannot handle it I make an SD card even though printing from MC is much nicer. Also the file slized fine with same parameters at version 2.0.0.10028 (or 26 whatever the last stable was) on the T100 and printed without a hitch.

martinpirringer, 2018-11-12:
One difference between this version and previous ones is that it seems after every "Wall Outer" and at some other places there are commands
G1 E nnnn

Sometimes multiple in a row whereas E nnn was usually and still is part of each G1 Xnn Ynn etc line. I printed a calibration box from MC and no noise then I exported the GCODE file and printed it from the SD and there was the noise and "dump" of filament

martinpirringer, 2018-11-12:
Update I will upload the video on the "END" - "FAIL" of the print, I restarted it. As it can't slice it on the T100 I sliced and exported a gcode file with the "include bed leveling" unchecked. and then loaded that gcode file into MC. Same part etc. It got past the point where it did the first "filament dump" without doing a dump. I also ran a gcode file for a calibration box and it too did the filament dump at the beginning and not when the gcode file w/o bedleveling gets run through MC. I tried it w/o bed leveling cause (i posted abug on that) if you save it with bed leveling and then run it through MC you get bed leveling x 2 which causes a mess. Seems it works like that this way now. I will let you know if that print finishes. 
[Gcode problem 2.zip](https://github.com/MatterHackers/MatterControl/files/2574090/Gcode.problem.2.zip)

Looking at the Gcode file w/o bed leveling that gets read into MC those lines referred to above with th G1 E nnnn are missing and I looked at it at the terminal flying by and they are not there. As far as I can tell they only appear if you save a Gcode file with bed leveling and if I read the code correct messes with the extruder. Hence it seemed as you can see on the video caused a net retraction of filament of at least 70 mm as I had to hit the 10mm E+ key at least 8 times before filament came out of the nozzle and that came out all right and its printing now so it wasn't a jam - so hope this helps. Well lets hope I don't waste more time and filament. Please let me know if I am on the right track - all I need to do is get this printed

jlewin, 2018-11-13:
Creating an internal issue to track the retraction bug experienced in the attached video. I'll report back when we know more. Thanks for report.

jlewin, 2018-11-13:
@martinpirringer  - so far I've been unable to reproduce the retraction behavior you ran into. If you can export your settings from the machine where you **produced** the GCode, I'll see if I can reproduce with your configuration.

You'll find the export option here:
[image]

Like with the video, zip the file so GitHub will accept it and attach below.

Thanks again!

martinpirringer, 2018-11-13:
[Pulse C-232 (1).zip](https://github.com/MatterHackers/MatterControl/files/2578347/Pulse.C-232.1.zip)
I am not completely sure that those are the settings as after the last post I exported the file to GCODE without bedleveling and then ran that gcode file through the T100 to not have the SD problem and it printed fine that way. I looked at the file as I think I stated and it did not have the lots of 
G1 E nnnn
Lines usually right at the begin of the "Outer Wall" Section and also some other places. I nailed down in the beginning on layer 0 you had multiple G1 Ennn lines in sequence and those were the ones there that made the filament "Dump" onto the print.