MatterHackers/MatterControl #4007: Cloud Library save does not work for updating

martinpirringer, 2018-11-19:
Bug Report
============
**Steps to Reproduce**
1. Load any STL
2. Click Save AS
3. Save to cloud Library
4. Make a change (either to STL or just move it into different position
5.) Click Save  AS again - and go to cloud library when the window pops up
6.) Assume you want to save it to the same name
7,) Click on the file you want to save it as just like in Windows
8,) You now have the changed copy and the old copy on your work plate.
**Expected Behavior**
The name in step 7 should pop into the name field so you can overwrite the existing file the only way to do an update ATM that I can figure is to delete the old file first then save the new version and retype the name (hopefully correctly)

**Actual Behavior**

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.0.0.10069
**Operating System Version**
Windows 10
**Printer Make/Model**
Pulse c232

jlewin, 2018-11-21:
CloudLibrary currently supports duplicate names with the same container. It was a design choose and as such, we don't overwrite and always create new. 

To edit an existing Cloud Library item you currently have to choose `Edit`, from either the right click or the overflow menus.

larsbrubaker, 2022-03-03:
fixde

closed 2022-03-03 by larsbrubaker