MatterHackers/MatterControl #4024: Layer stuck at 99.9% after printing is 100%, Printer doesn't turn off or rehome

Justin-Kirkman, 2018-11-27:
Current Build 2.0.0.10065

After the print job runs, the layers are stuck at 99.9% and never finishes resulting in the printer never turning off after the print run or going back to home.

[image]

This will leave the extruder and bed temperature still hot and on the corner of the print. 

[image]

Justin-Kirkman, 2018-11-27:
[image]

jlewin, 2018-11-27:
@larsbrubaker - terminal shows Connection stuck in G92 loop. Is this by chance a regression due to the recent G92 changes?

Justin-Kirkman, 2018-11-27:
[Pulse C-232 New.zip](https://github.com/MatterHackers/MatterControl/files/2621745/Pulse.C-232.New.zip)

jlewin, 2018-11-27:
Looks like the behavior is the result of #4641 where checksum is computed on a very long line in MatterControl and truncated in the printer firmware . Resetting End GCode is likely to resolve the problem and going forward we'll ensure that line lengths are limited to a sane value

closed 2018-12-07 by larsbrubaker