MatterHackers/MatterControl #4026: Brim outside build boundry

MichaelPetitclerc, 2018-11-28:
Bug Report
============
**Steps to Reproduce**
1.  Load a block into the build space that is the same volume as the build space
2.  Slice with Brim enabled
3.  Brim runs outside the print volume, no error is flashed on screen. 

**Expected Behavior**
Error to flash or not allow for the slice to complete
**Actual Behavior**
Slice completes and print can be run
**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")

**Operating System Version**
win 10, current build of MC 2.0 (2.0.0.10065)
**Printer Make/Model**
In slicer, across different printer profiles

Erudition, 2021-07-22:
Same with skirt.

Erudition, 2021-07-22:
This has caused printer errors and incorrect placement, so I hope this can be prioritized. In other engines, sizing a part on the bed automatically takes into account the brim and skirt settings so that if the total combination can't fit, the object appears ghosted.

In mattercontrol, the part will appear ghosted right up until you barely fit the object alone within the build volume - no extras included. You have to go to slice view to see the skirt and guess (!) if it goes out of bounds -- then come back and resize, then reslice and repeat. This is not a great UX.

More seriously, if you don't switch views, and just hit print, Mattercontrol will happily generate invalid gcode. The printer will be instructed to move outside the print area and can cause damage... 
In my case, the printer kept the head in-boundary when this happened, causing the output shape to be shifted and a corrupted mess.