MatterHackers/MatterControl #4047: MC2.0 - Overextruded First Layer with Large (1.0 or Larger) Nozzles

WindDrake, 2018-12-03:
Bug Report
============
**Steps to Reproduce**
1. Set up printer as 1.0mm Nozzle, any layer height.
2. Slice & Print
3. Examine Extrusion

**Expected Behavior**
Layer should be flat and extruded properly.

**Actual Behavior**
Layer is massively overextruded, causing high ridges and material to stick to the heatblock.

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
MC 2.0.0.10065

**Operating System Version**
Win7 x64 Pro

**Printer Make/Model**
Custom X/Y Gantry using Marlin 1.1.9 & E3D Titan w/ E3D Volcano w/ 1.0mm Nozzle

I've been able to fix this issue by using M221 to reduce the flow rate for layer 0 to 75%, then return it to 100% for layer 1. If I don't return it to 100%, layers 1 onward get underextruded, but manually editing the gcode is a chore every time.

unlimitedbacon, 2018-12-12:
I've done a lot of printing myself using larger nozzles with MatterControl and I've noticed the same phenomenon. I think that the slicer is working correctly. I have analyzed the gcode and could not find anything wrong with it.

I think that this is simply due to the nozzle being too low for the first layer. If you slice at 0.4 mm layers, but the nozzle is actually 0.2 mm off the bed, then the first layer will seem to be overextruded by 200%. The next few layers may also be overextruded, to a lesser degree.

The best solution is obviously to adjust your nozzle height, either physically or by using the Z Offset setting (Printer > General > Bed > Z Offset). Another workaround is to adjust the first layer extrusion multiplier (Slice Settings > Filament > Advanced > First Layer). You can set that to 75% or whatever you think is appropriate.

WindDrake, 2018-12-13:
The Z-Height (0.8mm Layer in this case) was verified with gauges, so sadly that's not it.

This isn't happening with a 0.4 or 0.6mm nozzle, either. I have yet to test an 0.8 as this printer is in heavy use with it's 1.0 nozzle.

blitzaxt, 2018-12-14:
I am seeing a similar issue with a 0.8mm nozzle. Just throwing my hat in the ring too.

WindDrake, 2018-12-18:
[Printer Profile.zip](https://github.com/MatterHackers/MatterControl/files/2690739/Printer.Profile.zip)

Done.

larsbrubaker, 2018-12-22:
@unlimitedbacon, can you look at this when you get a chance?

unlimitedbacon, 2018-12-26:
I cannot find anything wrong here.

Here is a test cube sliced with @WindDrake's settings. The layer view (with extrusion enabled) shows that the first layer is being printed correctly. The lines are the same width as the subsequent layers.

[image]

I also analyzed the G-Code and verified the calculated extrusion amounts. Here is a move from the first layer of the cube.
```
G1 X217.08 E36.762 F1800
G1 X218.67 E37.257
```
MatterSlice makes the (debatable) assumption that the extruded lines are rectangular. This means that the volume of plastic extruded is calculated like this.
```
Volume = length of move * nozzle diameter * layer thickness
```
From there, you get the amount to extrude by dividing the volume by the cross sectional area of the filament.
```
distance = Volume / ( pi * (filament diameter / 2) ^2 )
```
So here is the math for this particular move.
```
Volume   = (218.67 - 217.08) mm * 1.0 mm * 0.8 mm * 0.94
         = 1.272 mm^3
Distance = 1.19568 mm^3 / ( pi * (1.0 mm / 2) ^2 )
         = 0.529 mm
Distance * extrusion multiplier = 0.529 mm * 0.94
                                = 0.497 mm
```
Which is nearly the same result as in the actual G-Code.
```
E1 - E0 = 37.257 mm - 36.762 mm
            = 0.495 mm
```

WindDrake, 2018-12-28:
Just a follow-up on this.

As it stands the overextrusion issue started happening on my 0.4mm nozzle after some mechanical work to the printer's bed, which caused me to dig harder into this issue.

The M420 command (Leveling fade) introduced earlier this year in Marlin looks to have been at fault the whole time. The printer was moved to Marlin 1.1.9 from 1.1.5 around the same time MC 2.0 came out. At the time, we didn't know that levelling fade was enabled by default (or was even an option). 

Since we weren't giving the printer a fade height, the printer was not using it's collected mesh to level the build plate. As the machine had been manually leveled, this didn't come out as a problem until some work had been done to the bed.

We disabled leveling fade in firmware and the printer now seems to be behaving itself again. I haven't put the Volcano back on yet (this printer is generally going 24/7), but I'm hoping we'll have the same positive result.

Looks like this may have been a firmware issue, not a Mattercontrol issue. Sorry about that.