MatterHackers/MatterControl #4065: Support Material Does not Print

NorthRobin, 2018-12-06:
Bug Report
============
**Steps to Reproduce**
1. Open an STL file in MC
2. Slice, and print. 

**Expected Behavior**
The STL file should be sliced and include support material.
 
**Actual Behavior**
The file slices, but no support material shows up on the screen, nor does it print. The really strange thing is that the print head does the moves  where the support material should go, but the extruder stops completely.

**MatterControl Build Number**
V 2.0.0.10065

**Operating System Version**
W7 32 bit
**Printer Make/Model**
Rostock Max V2

[does not print support material.zip](https://github.com/MatterHackers/MatterControl/files/2653231/does.not.print.support.material.zip)

larsbrubaker, 2018-12-07:
Could you please attach (in a zip file) the profile and the part you are trying to print.

[image]

NorthRobin, 2018-12-08:
Hi Lars,

Thanks for looking at this. My printer is at my work, and I won't get back
there until Monday, but I will send this to you when I get in.

Thanks,

NorthRobin, 2018-12-10:
Hi Lars,

I forgot when I wrote to you on Friday that I had gone back to an earlier
version of  MC. When I have a moment, I will reinstall V2.x and get the
code to you. Thank you for your help.

NorthRobin, 2018-12-12:
Hi Lars,

Here are the files.

Thanks,

larsbrubaker, 2018-12-12:
I think you may have attached the files to your email and then replied. GitHub does not seem to accept attachments that way. You need to open in the issue in GitHub (available from the email) and then attache the files.

Thanks.

NorthRobin, 2018-12-12:
My apologies Lars. Her are the files
[Printer Settings and drawing.zip](https://github.com/MatterHackers/MatterControl/files/2672530/Printer.Settings.and.drawing.zip)

NorthRobin, 2018-12-18:
Hi All, 
I had some time today to check things out, and I uninstalled both versions of MC I had reinstalled MC 1.76 alongside of MC 2 to see if things still worked in the older version. To my surprise the same part was doing the same thing in the older version as well, even though it printed the same file with support prior to installing V2. I signed out of my MC account, and now it will create the support. As soon as I sign in, however, it will not print support. I'm thinking my printer profile must be corrupted in the MC cloud. I did check to make sure "print support material" was checked in the profile. I hope this information is somewhat helpful. 

For now I will stay signed out of my account, and copy the printer settings over to this installation manually. I will also see if not being signed into my account will solve the issues in V2 as well.

Thanks in advance for your time.

NorthRobin, 2018-12-18:
I just tried MC 2 without being signed into my account and it will produce support material. So I guess this is not a problem with MC2, although it only started once I installed the new version.

larsbrubaker, 2019-01-08:
We are going to be re-doing the support generation in the next build and it should improve some of what is happening here.

That said there are many things that might be a problem here.

1. the model is rotated so that a lot of it is in the air.
1. the model is not flat on the bottom
1. Air Gap has been turned to .2. This will make the support material impossible to remove (it defaults to .6)
1. Pattern Spacing has been changed to .5, this make very dense support and will use a lot of material (the default was 2.5).

I hope this helps and it sounds like you have some type of working solution. I don't see what else is happening here, yet, but I'll look some more.

NorthRobin, 2019-01-08:
Hi Lars,

Thank you very much for your time and advice. 

I did flip the part so that the "top" was flat on the bed, minimizing the support material.  I will check the other settings you mentioned. Over the holidays I spent quite a bit of time going through my settings and doing a lot of printing (I even printed my christmas gift tags;-0)

I just checked my settings, and the air gap and fill density were returned to the default settings, and things are going very well now. I really think this might have just been a one time glitch. I've since been able to stay logged into my account and everything works perfectly. The support material is generated properly.

Thank you again for the help. It is very much appreciated.

larsbrubaker, 2019-01-09:
You are very welcome.

closed 2019-01-09 by larsbrubaker