MatterHackers/MatterControl #4066: True Type font glyphs have error on some characters

larsbrubaker, 2018-12-06:
@larsbrubaker commented on [Mon Dec 03 2018](https://github.com/MatterHackers/MCCentral/issues/4659)

[image]

This seems to be a parsing bug in IGlyphReaderExtensions. The 'O' in particular looks like it has an error figuring out if it is in a curve or not and renders the LineTo that seems out of place.

@prepare, any insight or suggestions you might have in how to fix this would be incredibly appreciated.

larsbrubaker, 2018-12-06:
Put up a bounty on this bug as a test. First time trying this out.

https://www.bountysource.com/issues/67138827-true-type-font-glyphs-have-error-on-some-characters

prepare, 2018-12-07:
hello, @larsbrubaker 

Could you provide that font ?

larsbrubaker, 2018-12-07:
It can be found in:
MatterControl\StaticData\FontsAlfa_Slab.ttf

prepare, 2018-12-07:
The bug is fixing , and I will send a PR soon.

[image]
_pic 1: before fixing, FontsAlfa_Slab.ttf_

---
[image]

_pic 2: after_

Thank you for your report :)

prepare, 2018-12-07:
This PR => https://github.com/MatterHackers/agg-sharp/pull/1112 will fix this problem

closed 2018-12-08 by larsbrubaker

larsbrubaker, 2018-12-08:
@prepare you should claim the bounty in bountysource. It is only 15$ but you still did the work and it is very helpful.

prepare, 2018-12-09:
@larsbrubaker ,
Thank you for your kindness.

At this time, I can't accept the bounty,

But Feel free to send any bugs/ suggestion to me.