MatterHackers/MatterControl #4068: Can't connect over USB on a Mac.

8perf, 2018-12-06:
Bug Report
============
**Steps to Reproduce**
1. Start MC
2. Turn on printer
3. add printer in MC - Prusa Research i3 MK2
4. follow prompts to set up over USB - optional to do a manual set up and select the USBtoUART driver as its the only applicable one on the list.  NOTE:  Previous driver suggestions from older versions of MC already installed in the past but do not show up in the list. Never got older versions of MC to work either.

**Expected Behavior**
Positive connection and set up notice
**Actual Behavior**
Printer beeps as it normally would with a successful connection in Windows but line of red text pops up and says "Oops.  Printer could not be detected."

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
version 2.0.0.10074

**Operating System Version**
Mac OS 10.12.6

**Printer Make/Model**
Anycubic i3  Prusa clone

***** Printer works just fine with this driver/ setup under Windows 10 on a laptop*****

jlewin, 2018-12-17:
@8perf  - Can you connect to your printer on your Mac with other printing hosts? Occasionally on a Mac getting certain FTDI drivers setup for clone hardware is challenging, if you're having success with other software on the Mac that might isolate the problem to us.

8perf, 2018-12-18:
I can connect with a modified Cura version that came with the printer, (15.04.6 I think) and Craftware, but both crash when I try to print anything.

boatgunnel, 2018-12-27:
I am having the exact same issue.  I bought a Robo R1 a few years ago.  I have an old laptop running Windows 7.  The laptop is ancient.  I recently got a new imac for the shop, and was hoping to use it strictly for my 3d printer and CNC machine, as well as design work.  I hooked the printer up and it would say connected in the software, but the printer would not respond to any commands at all.  I have tried repeatedly changing versions of the software to no avail.  Went back to the windows laptop and it connected and responded as soon as it was plugged in.  I am in desperate need of help as i am not sure how much longer the windows laptop will last.  Thanks.

boatgunnel, 2019-01-23:
> Connecting a printer to a Mac almost always just works... except when it doesn't. If a stock OSX setup fails to connect to a printer and/or if the port doesn't show up, then getting the right drivers installed can be a real headache. MatterControl is at the mercy of the operating system when it comes to serial connections. The fastest way to identify where to spend your time troubleshooting is to determine if other applications can connect to the printer. If Cura or RepHost connect without issue but we do not, then certainly something is wrong with our setup and requires attention.
>
> Otherwise you might have to dig in and work through chipset/OS/driver issues. In my experience, this often happens with FTDI/clone chipsets and in the past I've used the articles below to get finicky OSX machines working:
>
> http://forum.arduino.cc/index.php?topic=198539.0
> http://www.ftdichip.com/Support/Documents/AppNotes/AN_134_FTDI_Drivers_Installation_Guide_for_MAC_OSX.pdf
> https://learn.sparkfun.com/tutorials/how-to-install-ftdi-drivers/mac

Ok well I fixed this problem. The iMac usb power was on the low side. I hooked up a usb hub and it started working. Now on the offset in mattercontrol. How do I get the printer off the bed during the first pass?? I want it to raise a little. I looked through the instructions in the matter control message boards but still couldn’t figure it out. Thanks 

Jordan Johnson

jlewin, 2019-01-23:
Great to hear you found a resolution.

Github isn't the best place to answer support questions but in general you can use bed leveling, z-offset and babystepping to tune the height of the nozzle, each with their own trade-offs. Hopefully that's enough to point you in the right direction

bmwertman, 2019-07-22:
I'm experiencing the same thing with my FLSUN i3. I tried manually connecting under both the Prusa MK2 profile and the one under Other -> Prusa i3l. When I try manually my LCD screen shows Marlin rebooting when I click connect but then I get the following error message.

[image]

eeyore124, 2020-01-26:
I'm having similar problem with my MacBook.  I can connect from a windows machine.  I can connect to a Arduino with the MacBook.  I would like to use my MacBook with the 3D printer.