MatterHackers/MatterControl #4090: Skirt issue with Raft

blueeagle69, 2018-12-17:
Bug Report
============
**Steps to Reproduce**
1. Use Skirt to prime nozzle
2. Use Raft for adhesion
3. 

**Expected Behavior**
Skirt should print, move onto the Raft and start printing the object
**Actual Behavior**
Skirt prints, followed by the raft. But after the Raft is completed the printer then decides to go over the existing Skirt again, but at its current height. Surely the printer wouldn't need to print the Skirt again? Let-alone at the height of the raft's top layer.
I was printing from SD card.
**MatterControl Build Number**
2.0.0.10065

Windows 10

Anet A8, Marlin 1.1.9

larsbrubaker, 2018-12-17:
Thanks for the report. This has been fixed and will be in the next release.

Coming very soon.

closed 2018-12-17 by larsbrubaker

blueeagle69, 2018-12-17:
Cheers Lars!
Merry Christmas :)