MatterHackers/MatterControl #4092: Second Text Subtraction fails to complete

HmBeerBrewer, 2018-12-18:
Bug Report
============
**Steps to Reproduce**
1. Create a cube that is 50mm wide, 18mm deep and 4 mm high.
2. Create a text item that says "Agenda", accept all text defaults, make it 1mm high and raise it 3mm off the bed.
3. Center text item on the box and subtract.  This part works fine.
4. Flip the box along the long axis 180 degrees.
5. Create text item that says "Done", accept all text defaults, make it 1 mm high and raise it 3mm off the bed.
6. Center the text item on the box and subtract.  It will only subtract the first letter or two.

**Expected Behavior**
All letters should be subtracted when the second text time is subtracted.

**Actual Behavior**
Only the first letter or two are subtracted, the rest remain filled in.  I've tried this about 8 times now. It only worked once out of the 8 attempts.  I've tried with different fonts with the same results.

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.0.0.10065

**Operating System Version**
Windows 10 Build 1809, latest OS patches installed.

**Printer Make/Model**
Anet A8.
[image]

larsbrubaker, 2018-12-18:
Thank you for the bug report. This is fixed in source and will be in the next release (coming soon).

We have introduced a much better boolean engine and these problems should be much less frequent.

[image]

HmBeerBrewer, 2018-12-19:
I look forward to trying the new version.  Closing this bug report.

closed 2018-12-19 by HmBeerBrewer