MatterHackers/MatterControl #4129: Pulse XE SD Card print freezes at end of print and keeps extruding a glob of PLA.

bsetter, 2018-12-29:
At the end of a print from an SD card the Pulse XE freezes on the end of the last layer (right when the printer would normally "home" and send the print bed forward with the final printed object) but instead it freezes and keeps extruding making a big glob of PLA on the printed object. It's as-if the printer doesn't know what to do when it finishes the g-code print. This issue seems to only happen when printing from G-Code from an SD card.

larsbrubaker, 2019-01-04:
This should be fixed in the latest build (made available just today).

Please let us know if you continue to see this or any other bugs.

closed 2019-01-04 by larsbrubaker

RSteel517, 2019-01-05:
So for, the last few quick test prints I have done, I have not experienced this again.

If it happens again I'll let ya'll know and will send over the G-Code file.

Thanks for fixing this!

bsetter, 2019-01-06:
I did a few test prints and here's the file I used using Matter Control Build 2.0.0.10074 issue still exists where end of print causes all axis freeze and extrusion to continue creating a glob on the object and potential hazard cause I have to stop the printer manually when it happens.
[testcuberyno.gcode.zip](https://github.com/MatterHackers/MatterControl/files/2730943/testcuberyno.gcode.zip)

larsbrubaker, 2019-01-07:
@bsetter, this issue has been fixed in source and we have promoted a Windows build with the fix. We will get a Mac version together this next week and release it to Alpha. It will then progress through release as we get feedback (to make sure the build is solid) and should be public in 2 to 3 weeks from now.

You can switch to the alpha channel if you wish to see the build when it goes up this week:

[image]

bsetter, 2019-01-07:
Thanks!

<https://user-images.githubusercontent.com/1158332/50743572-6d86d980-11ce-11e9-8bbc-8947169231c3.png>