MatterHackers/MatterControl #4154: Cloud Sync

MichaelPetitclerc, 2019-01-07:
Bug Report
============
**Steps to Reproduce**
1.  Upload a model to Cloud Sync
2. Select model to print

**Expected Behavior**
Print job is started and you are notified that it has
**Actual Behavior**
Cloud Sync said the Mattercontrol version is not compatible and must be of 1.4.2 or higher even when using MC 2.19.1
**MatterControl Build Number**
2.19.1.10116
**Operating System Version**
Win 10 x64

**Printer Make/Model**
Pulse C232 or C233

jlewin, 2019-01-08:
@MichaelPetitclerc - ping me when you have a second and could possibly show this in action

jlewin, 2019-01-11:
[image]

jlewin, 2019-01-11:
This issue is caused by a failure in MatterControl to pass the `SyncFeatureSet` param on the `update-device-status` requests. The regression occurred in MatterHackers/MCCentral@fcf4563 when P2P params were removed and three unintended items were dropped (`SyncFeatureSet, ClientHasCamera, ClientName`)

closed 2019-01-18 by larsbrubaker