MatterHackers/MatterControl #4155: Unhandled exception error on opengl32.dll while loading Gcode

Regnad517, 2019-01-08:
Bug Report
============
**Steps to Reproduce**
1. Whenever I Slice it crashes at loading Gcode
2. I did a clean install of Windows and it printed great for 1 day.  The next day I went to print and I get the exception error in the pic attached.  When the error pops up, I am frozen completely unless I quit.
3. I have not modified any of the default settings or gcode to reproduce this error.  I have even done a complete scrub and reloaded MC and still crashes. 

**Expected Behavior**

**Actual Behavior**

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
V2.19.1.10116
**Operating System Version**
Win7 64bit
**Printer Make/Model**
Monoprice Maker Select V2
[image]

Regnad517, 2019-01-08:
So I have tried it from 2 different Win7 computers and same exact results.  The computer I use for this is dedicated so I can make any changes at any time.  Going to try and load Win10 on it to see if the problem still persists.

Any info I can add to help resolve, feel free to ask me.

Thanks

Regnad517, 2019-01-26:
I have to say this is getting very annoying having to completely erase my machine and install Windows, MC, etc... every 2 days.  And not a single response here at all.  It's too bad, I love using MC and being local in Lake Forest, visiting the store but hate being ignored.  I guess you can close this thread, I will give my business to someone that wants it.

larsbrubaker, 2019-01-26:
I apologize that you are feeling ignored. 

GitHub is not a support forum. Please contact MatterHackers support if you need support. 

This site is for the developers of MatterControl. 
They take the issues that are reported here and then triage them into various development cycles. They do not necessarily respond to any issue, or commit to when they will be addressed.

We very much appreciate you reporting issues, but our developers are not in a position to offer individual support.

larsbrubaker, 2019-01-27:
Again, sorry for the misunderstanding and thank you for the additional information. 

We will still be investigating this when we have the time. It is very useful for us for you to report these kinds of issues.

I hope to encourage you to continue to report issues. We just don't have the capability to give individual support to issues, or prioritize them above our development schedule.

compuwar, 2021-04-04:
I am having the same issue running MC from an RDP session on Win10 Pro.  If I switch to 2D mode, the crash does not happen.

- <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event"> 
- <System> 
<Provider Name=".NET Runtime" /> 
<EventID Qualifiers="0">1026</EventID> 
<Level>2</Level> 
<Task>0</Task> 
<Keywords>0x80000000000000</Keywords> 
<TimeCreated SystemTime="2021-04-04T21:17:18.081666800Z" /> 
<EventRecordID>1646</EventRecordID> 
<Channel>Application</Channel> 
<Computer>printserver</Computer> 
<Security /> 
</System> 
- <EventData> 
<Data>Application: MatterControl.exe Framework Version: v4.0.30319 Description: The process was terminated due to an unhandled exception. Exception Info: System.AccessViolationException at OpenTK.Graphics.OpenGL.GL+Core.DrawArrays(OpenTK.Graphics.OpenGL.BeginMode, Int32, Int32) at MatterHackers.Agg.UI.OpenTkGl.DrawArrays(MatterHackers.RenderOpenGl.OpenGl.BeginMode, Int32, Int32) at MatterHackers.RenderOpenGl.OpenGl.GL.DrawArrays(MatterHackers.RenderOpenGl.OpenGl.BeginMode, Int32, Int32) at MatterHackers.GCodeVisualizer.GCodeVertexBuffer.RenderTriangles(Int32, Int32) at MatterHackers.GCodeVisualizer.GCodeRenderer.Render3D(MatterHackers.GCodeVisualizer.GCodeRenderInfo, MatterHackers.Agg.UI.DrawEventArgs) at MatterHackers.MatterControl.BedConfig.RenderGCode3D(MatterHackers.Agg.UI.DrawEventArgs) at MatterHackers.MatterControl.PartPreviewWindow.View3DWidget.MatterHackers.MatterControl.PartPreviewWindow.IDrawable.Draw(MatterHackers.Agg.UI.GuiWidget, MatterHackers.Agg.UI.DrawEventArgs, MatterHackers.VectorMath.Matrix4X4, MatterHackers.VectorMath.WorldView) at MatterHackers.MatterControl.PartPreviewWindow.Object3DControlsLayer.DrawGlContent(MatterHackers.Agg.UI.DrawEventArgs) at MatterHackers.MatterControl.PartPreviewWindow.Object3DControlsLayer.RenderSource_BeforeDraw(System.Object, MatterHackers.Agg.UI.DrawEventArgs) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.MatterControl.PartPreviewWindow.PrinterTabPage.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.GuiWidget.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.MatterControl.RootSystemWindow.OnDraw(MatterHackers.Agg.Graphics2D) at MatterHackers.Agg.UI.WinformsSystemWindow.OnPaint(System.Windows.Forms.PaintEventArgs) at MatterHackers.Agg.UI.OpenGLSystemWindow.OnPaint(System.Windows.Forms.PaintEventArgs) at System.Windows.Forms.Control.PaintWithErrorHandling(System.Windows.Forms.PaintEventArgs, Int16) at System.Windows.Forms.Control.WmPaint(System.Windows.Forms.Message ByRef) at System.Windows.Forms.Control.WndProc(System.Windows.Forms.Message ByRef) at System.Windows.Forms.Form.WndProc(System.Windows.Forms.Message ByRef) at System.Windows.Forms.NativeWindow.Callback(IntPtr, Int32, IntPtr, IntPtr)</Data> 
</EventData> 
</Event>

<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event"> 
- <System> 
<Provider Name="Application Error" /> 
<EventID Qualifiers="0">1000</EventID> 
<Level>2</Level> 
<Task>100</Task> 
<Keywords>0x80000000000000</Keywords> 
<TimeCreated SystemTime="2021-04-04T21:17:21.002944000Z" /> 
<EventRecordID>1647</EventRecordID> 
<Channel>Application</Channel> 
<Computer>printserver</Computer> 
<Security /> 
</System> 
- <EventData> 
<Data>MatterControl.exe</Data> 
<Data>1.0.0.0</Data> 
<Data>606765e7</Data> 
<Data>OPENGL32.DLL</Data> 
<Data>10.0.18362.387</Data> 
<Data>c7cf39ed</Data> 
<Data>c0000005</Data> 
<Data>000000000001b919</Data> 
<Data>3dbc</Data> 
<Data>01d7299748915167</Data> 
<Data>C:\Program Files (x86)\MatterControl\MatterControl.exe</Data> 
<Data>C:\WINDOWS\SYSTEM32\OPENGL32.DLL</Data> 
<Data>675e5db7-3d5b-40cf-80f6-7312fd842be6</Data> 
<Data /> 
<Data /> 
</EventData>

<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event"> 
- <System> 
<Provider Name="Windows Error Reporting" /> 
<EventID Qualifiers="0">1001</EventID> 
<Level>4</Level> 
<Task>0</Task> 
<Keywords>0x80000000000000</Keywords> 
<TimeCreated SystemTime="2021-04-04T21:17:28.065442100Z" /> 
<EventRecordID>1648</EventRecordID> 
<Channel>Application</Channel> 
<Computer>printserver</Computer> 
<Security /> 
</System> 
- <EventData> 
<Data>1370708196825226691</Data> 
<Data>4</Data> 
<Data>APPCRASH</Data> 
<Data>Not available</Data> 
<Data>0</Data> 
[... 26 lines cut ...]
<Keywords>0x80000000000000</Keywords> 
<TimeCreated SystemTime="2021-04-04T21:17:28.737403500Z" /> 
<EventRecordID>1649</EventRecordID> 
<Channel>Application</Channel> 
<Computer>printserver</Computer> 
<Security /> 
</System> 
- <EventData> 
<Data>MatterControl.exe</Data> 
<Data>1.0.0.0</Data> 
<Data>606765e7</Data> 
<Data>OPENGL32.DLL</Data> 
<Data>10.0.18362.387</Data> 
<Data>c7cf39ed</Data> 
<Data>c000041d</Data> 
<Data>000000000001b919</Data> 
<Data>3dbc</Data> 
<Data>01d7299748915167</Data> 
<Data>C:\Program Files (x86)\MatterControl\MatterControl.exe</Data> 
<Data>C:\WINDOWS\SYSTEM32\OPENGL32.DLL</Data> 
<Data>5d180804-5df6-4e8f-9433-a7f473c63f72</Data> 
<Data /> 
<Data /> 
</EventData> 
</Event>

compuwar, 2021-05-30:
> Again, sorry for the misunderstanding and thank you for the additional information.
>
> We will still be investigating this when we have the time. It is very useful for us for you to report these kinds [...]

Seems like two years would be an excessive time not to address a bug.  What then is the preferred reporting procedure?  My guess is the memory violation here affects more that this instance, and is why I couldn’t get OS X working consistently either.