MatterHackers/MatterControl #4164: Mattercontrol uses incorrect layer width when using different 1st layer height.

DEJDesign, 2019-01-12:
Bug Report
============
**Steps to Reproduce**
1. Open 3DBenchy model in MatterControl.
2. Select "Fine" for quality and "Pro Series PLA" for filament in slicer settings.
3. Look at the 1st layer of the sliced model and compare to the rest of the model...the entire first layer has the wrong width and looks over extruded.

**Expected Behavior**
Mattercontrol should use a consistent width for the extrusion path, even when using a different layer height.

**Actual Behavior**
The extrusion width for the 1st layer is larger than it should be, resulting in overextrusion of the entire 1st layer and creating a small brim on the sliced model.

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.0.0.10074

**Operating System Version**
MacOS 10.14.2

**Printer Make/Model**
Pulse XE

[image]
[image]

larsbrubaker, 2019-01-13:
Thank you for the report. This is an error in the state height of the layer in the gcode hints. The actual gcode should be correct and the print should work fine.

The code has been updated and tested and will produce the correct results in the next release (probably March)

Here is the new code running.

[image]

[image]

You can see the bottom layer is the correct height at .25, and the correct extrusion width.

closed 2019-01-18 by larsbrubaker

DEJDesign, 2019-02-17:
This issue has not been fixed and is still an issue in the 2.19.1.10123 version of MC2.0 for MacOS.