MatterHackers/MatterControl #4173: Additional support options.

DEJDesign, 2019-01-13:
Feature Request
===============

**Problem**
Unless the surfaces are horizontal, MatterControl often fails to detect steep overhanging sections that require support material, resulting in print failure and/or severe warping.

[image]

**Solution**
1. Give users control over the max overhang angle that will be tolerated without support.
2. Include additional support types (accordion-style) that are easier to remove than the existing LINES and GRID types.
3. Give users the ability to remove sections of the auto-generated support where it isn't wanted (similar to the way Simplify 3D lets users delete and add support pillars) without having to create custom supports from primitives.

[image]
[image]
[image]

larsbrubaker, 2019-01-14:
Support pillars will be in the next version of MatterControl (likely released in February)

DEJDesign, 2019-01-14:
Thanks for the update Lars! Will any consideration be given to a "max overhang angle" option to allow users to trouble shoot support not forming under steep overhangs as shown in the example above?

unlimitedbacon, 2019-01-17:
@DEJDesign, I think the setting you are looking for is [Support Percent](https://www.matterhackers.com/mattercontrol/support/slice-settings/support/support). This is equivalent to the support angle setting in other slicers.

closed 2019-01-18 by larsbrubaker