MatterHackers/MatterControl #4181: Bed wireing

martinpirringer, 2019-01-14:
Question
========

This issue tracker is for our MatterControl development team and they are not in a position to answer support questions.

If you have a question about using MatterControl, please ask it on the MatterControl forums.
https://forums.matterhackers.com/category/20/mattercontrol

martinpirringer, 2019-01-14:
Posting it here as the email said it exceeds some size limit . This was the email
[Bedwiring.zip](https://github.com/MatterHackers/MatterControl/files/2753835/Bedwiring.zip)
Hi,

The good part is that you zip tied the wire to the electronics case to
not put constant strain on the plug and wiggle it loose. The bed part is
that I cannot print in the back corner as the wire encroaches into the
build volume. AND! its kinking already cause it is too short. IDK what
the amperage is on the bed but the connectors according to prusa which
also uses a rambo mini board can take all the way to 12 gauge wires. I'd
go 12 gauge with silicon insulation (a nice and flexible one) This wire
does not look like even 14 gauge to me it looks more like 16 or 18. But
am not touching it to find out while its working. It really needs a drag
chain - see video. Either that or put on the website "use only for small
parts the size of the test cube or Astronaut filament"

Martin

larsbrubaker, 2019-01-14:
This is confusing. Our support team does not use GitHub to track errors or hardware issues. I do not know that they have a way to look at this, but this is not the right place to add specific support issues.

Please work with support to handle this issue in their system.

closed 2019-01-17 by unlimitedbacon