MatterHackers/MatterControl #4193: Mac version crashing

MichaelPetitclerc, 2019-01-15:
Bug Report
============
**Steps to Reproduce**
1) The Mac version download (2.0.0.10074, running on Mojave 10.14.2) 
drops dead the instant I try to save anything. No message, no dialog, no 
crash report. It simply disappears from the screen, and nothing has been 
saved. I did create an account and log in (see #2), but to no avail. 
Same result if saving to Downloads or to Library.

**Expected Behavior**

**Actual Behavior**

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.0.0.10074
**Operating System Version**
Mac Mojave 10.14.2
**Printer Make/Model**

hashfunktion, 2019-03-25:
I have exactly the same problem. :raised_hand:

ErikEscudero, 2019-07-30:
Comment from our MC forum:

MC is crashing on every export option I choose.
(also items saved in cloud library are showing up as as selection but once downloaded it's an empty object).
I have an object on my iMac i want to get on to my MacBook that I use to connect to my Rep 2X but so far I'm being confounded at every turn :-(.

Crash report from iMac.
Last login: Thu Jul 4 14:38:52 on console
iMac:~ andrewodowd$ /Applications/MatterControl.app/Contents/MacOS/MatterControlMac ; exit;
Time to '': 333
Time to 'ShowAsSystemWindow': 0
Time to 'First draw->RunOnIdle': 0
Time to 'MatterControlApplication.Initialize': 0
Time to 'Loading...': 0
Time to 'Loading...': 0
Time to 'Loading...': 0
Time to 'Loading...': 0
Time to 'Loading...': 116
Time to 'Loading...': 160
Time to 'Loading...': 152
Time to 'AddChild->MainView': 0
SendPrintEvent - AuthAndSessionValid: True; PrinterSelected: True; TokenPopulated: False
SendPrintEvent - AuthAndSessionValid: True; PrinterSelected: True; TokenPopulated: False
SendPrintEvent - AuthAndSessionValid: True; PrinterSelected: True; TokenPopulated: False
SendPrintEvent - AuthAndSessionValid: True; PrinterSelected: True; TokenPopulated: False
Connected to device channel: ag9zfm1hdHRlcmNvbnRyb2xyKQsSB1Byb2plY3QY7AcMCxIOQ2xpZW50Q29uc3VtZXIYgICQ19qPrQgM
Connected to user channel: agxzfmJydWhhLWF1dGhyGAsSC1VzZXJQcm9maWxlGICAgPnYqdwIDA

[ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object
at MatterHackers.MatterControl.ExportPrintItemPage+<>c__DisplayClass3_0.<.ctor>b__0 (System.Object s, MatterHackers.Agg.UI.MouseEventArgs e) [0x0005b] in :0
at MatterHackers.Agg.UI.GuiWidget.OnClick (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x0000a] in :0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00256] in :0
at MatterHackers.MatterControl.CustomWidgets.SimpleButton.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00000] in :0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in :0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in :0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in :0
at MatterHackers.Agg.UI.SystemWindow.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00017] in :0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in :0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in :0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in :0
at MatterHackers.Agg.UI.SystemWindow.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00017] in :0
at MatterControlMac.MacPlatformWindow.MouseUp (AppKit.NSEvent theEvent) [0x0004a] in :0
at (wrapper managed-to-native) AppKit.NSApplication.NSApplicationMain(int,string[])
at AppKit.NSApplication.Main (System.String[] args) [0x00040] in <3e1eecfdb2534f518bea4f1a21972847>:0
at MatterControlMac.MainClass.Main (System.String[] args) [0x00012] in :0
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...7 completed.

[Process completed]

ErikEscudero, 2019-08-09:
Comment in support email:

Steps to reproduce:
1. Drop Cube into workspace
2. Right click on Cube
3. Select 'export'
4. Select 'STL'
5. Click 'Export'
6. Boom! :)

Time to '': 286
Time to 'ShowAsSystemWindow': 0
Time to 'First draw->RunOnIdle': 0
Time to 'MatterControlApplication.Initialize': 0
Time to 'Loading...': 0
Time to 'Loading...': 0
Time to 'Loading...': 0
Time to 'Loading...': 0
Time to 'Loading...': 105
Time to 'Loading...': 150
Time to 'Loading...': 47
Time to 'AddChild->MainView': 0
SendPrintEvent - AuthAndSessionValid: False; PrinterSelected: True; TokenPopulated: True
SendPrintEvent - *Skipped* AuthAndSessionValid: False

[ERROR] FATAL UNHANDLED EXCEPTION: System.NullReferenceException: Object reference not set to an instance of an object
at MatterHackers.MatterControl.ExportPrintItemPage+<>c__DisplayClass3_0.<.ctor>b__0 (System.Object s, MatterHackers.Agg.UI.MouseEventArgs e) [0x0005b] in <ca1405d15e5d42b589cad20282092f68>:0
at MatterHackers.Agg.UI.GuiWidget.OnClick (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x0000a] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00256] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterHackers.MatterControl.CustomWidgets.SimpleButton.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00000] in <ca1405d15e5d42b589cad20282092f68>:0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterHackers.Agg.UI.SystemWindow.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00017] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterHackers.Agg.UI.GuiWidget.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00180] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterHackers.Agg.UI.SystemWindow.OnMouseUp (MatterHackers.Agg.UI.MouseEventArgs mouseEvent) [0x00017] in <f3a6caf3214c4942a581165cc935211a>:0
at MatterControlMac.MacPlatformWindow.MouseUp (AppKit.NSEvent theEvent) [0x0004a] in <f6b6c4b89de3427ab6b4c2f9608ae461>:0
at (wrapper managed-to-native) AppKit.NSApplication.NSApplicationMain(int,string[])
at AppKit.NSApplication.Main (System.String[] args) [0x00040] in <3e1eecfdb2534f518bea4f1a21972847>:0
at MatterControlMac.MainClass.Main (System.String[] args) [0x00012] in <f6b6c4b89de3427ab6b4c2f9608ae461>:0

ishapiro, 2019-09-04:
I am seeing the exact same issue.   This creates a catch 22.   I would like to teach my YouTube channel viewers about the new modeling tools in MatterControl.  However the slicer quality is not great, specifically this issue "Start/stop of inner perimeter loops is causing artifacts on the outside #344" makes the prints ugly.   So I tried using the image conversion tool to create a model and export it as an STL.   Unfortunately exporting STL files on a Mac crash.

Is there a workaround for either issue?

ckartchnermh, 2019-09-11:
Another user is experiencing this. Mac 10.14.6 and MC Version 2.19.7.10356

dalgleish, 2019-09-13:
I think I see how to fix this.
I'll try to get a PR out this weekend.

larsbrubaker, 2019-09-14:
If you build from branch 2.19.7 you will be on what we are currently working on an update to. Master has some work that is not ready yet in it. That said if you make a fix against master, we can easily incorporate it into 2.19.7.

Thanks in advance for your help.