MatterHackers/MatterControl #4275: View window scrolling

Profkanz, 2019-02-08:
Feature Request
===============
When Zooming in to a very close view, it would help if the view window could be scrolled up or down and right or left.
**Problem**
Describe the problem you feel needs to be solved.

**Solution**
Explain how this feature will solve the problem.

larsbrubaker, 2019-02-08:
You should be able to pan the window around with either the middle mouse button, or by holding the control+shift+left mouse button.

Please let me know if this is not the issue you are describing.

closed 2019-02-08 by larsbrubaker

Profkanz, 2019-02-09:
Those options behave like a slomo version of the right mouse. They do move the view but with a distant pivot point. I want to be able to move the view without changing the point of view.For example, sometimes I want to be in an extreme zoomed in view and then move the view around without changing the viewing angle. This is for a relatively complex part when I want to verify if the desired surface is located flat on the build plate.