MatterHackers/MatterControl #4278: Menu's missing in MC Alpha

MichaelPetitclerc, 2019-02-08:
Bug Report
============
**Steps to Reproduce**
Software is not showing the primitives or allowing the user to rotate the object.  Primitives menu shows up blank.

**Expected Behavior**
-
**Actual Behavior**
-
**MatterControl Build Number**
MC Alpha build 2.19.2.10154

**Operating System Version**
MacOS 10.11.6
**Printer Make/Model**
None, in the Design tab of MC
[image]

DEJDesign, 2019-02-09:
Having the same issue on MacOS 10.14.2

jlewin, 2019-02-27:
Problem with plugin registration were preventing plugins from loading. This is fixed in Experimental 
2.19.2.10172 and should make it into an Alpha release later this week

[image]

closed 2019-03-02 by larsbrubaker