MatterHackers/MatterControl #4281: Pulse D435 Infinity home G-Code

MichaelPetitclerc, 2019-02-12:
Bug Report
============
**Steps to Reproduce**
1. Slice a G-Code in the Pulse D435 profile.
2.  Export as G-Code and open the G-Code in notepad
3.  Line G1 reads "G1 XInfinity YInfinity Z-Infinity F1800"

**Expected Behavior**
Print head to go to the front left to begin purge line
**Actual Behavior**
Print head slowly moved to the back right corner and never stops when limit is hit. 
**MatterControl Build Number**
2.19.1.10116

**Operating System Version**
Win 10
**Printer Make/Model**
Pulse D435 Volcano
G-Code reference attached.
[Frame Guard 2ct (1).zip](https://github.com/MatterHackers/MatterControl/files/2857383/Frame.Guard.2ct.1.zip)

larsbrubaker, 2019-02-15:
Cannot reproduce and there is not enough data in this issue to reproduce.

If there is a profile and a part that produce this error, we can re-open and fix it.

jlewin, 2019-02-21:
@MichaelPetitclerc - were you able to get the .printer settings from the customer who was running into this error? I still believe if we export there settings and import them here, we'll be able to reproduce and determine the cause of this issue

closed 2019-04-15 by larsbrubaker