MatterHackers/MatterControl #4284: Matter Control not working on MAC OS Mojave 10.14.3

conicio, 2019-02-13:
Hi folks.
I'm a newbie at the 3d printing world, I've downloaded the last version of MatterControl and when I try to run it crashes to desktop. It's a MacBook Pro (Retina, 15 pulgadas, mids 2014) running MAC OS Mojave 10.14.3 The printer is an I3 Steel with RAMPS1.4+ARDUINO MEGA2560+4988 drivers, loaded with the last release of Marlin firmware. I can't tell you the build because I've never got to run the program.
If I run from terminal I get this:
/Applications/MatterControl.app/Contents/MacOS/MatterControlMac ; exit;
Time to '': 184
2019-02-12 14:11:34.449 MatterControlMac[67012:801670]
Unhandled Exception:
System.TypeInitializationException: The type initializer for 'MatterHackers.MatterControl.SlicerConfiguration.ProfileManager' threw an exception. ---> System.NullReferenceException: Object reference not set to an instance of an object
at MatterHackers.MatterControl.AppContext.get_Theme () [0x00000] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController.get_Theme () [0x00000] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController..ctor () [0x00098] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController.get_Instance () [0x00007] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.PrinterInfo.set_ID (System.String value) [0x00000] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.DataStorage.ClassicDB.ClassicSqlitePrinterProfiles.ImportPrinter (MatterHackers.MatterControl.DataStorage.Printer printer, MatterHackers.MatterControl.SlicerConfiguration.ProfileManager profileData, System.String profilePath) [0x00021] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.DataStorage.ClassicDB.ClassicSqlitePrinterProfiles.ImportPrinters (MatterHackers.MatterControl.SlicerConfiguration.ProfileManager profileData, System.String profilePath) [0x00028] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager.EnsurePrintersImported () [0x00026] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager.Load (System.String userName) [0x00052] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager.ReloadActiveUser () [0x0005b] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager..cctor () [0x00016] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
--- End of inner exception stack trace ---
at (wrapper managed-to-native) System.Object.__icall_wrapper_mono_generic_class_init(intptr)
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager.add_UserChanged (System.EventHandler value) [0x00000] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController..ctor () [0x0007d] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController.get_Instance () [0x00007] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.DialogPage..ctor (System.String cancelButtonText, System.Boolean useOverflowBar) [0x00038] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at LicenseAgreementPage..ctor (System.Action acceptAction) [0x0000d] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.Application+<>c__DisplayClass5_0.b__2 (System.Object s, System.EventArgs e) [0x00023] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.Agg.UI.GuiWidget.OnLoad (System.EventArgs args) [0x0000a] in <4d176e336ae140988af12ff0893dabe2>:0
at MatterHackers.Agg.UI.GuiWidget.OnDraw (MatterHackers.Agg.Graphics2D graphics2D) [0x00008] in <4d176e336ae140988af12ff0893dabe2>:0
at MatterHackers.MatterControl.RootSystemWindow.OnDraw (MatterHackers.Agg.Graphics2D graphics2D) [0x00021] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterControlMac.MacPlatformWindow.<.ctor>b__18_1 (System.Object s, OpenTK.FrameEventArgs e) [0x0004f] in :0
at OpenTK.Platform.MacOS.MonoMacGameView.OnRenderFrame (OpenTK.FrameEventArgs e) [0x00010] in <325fbbce19e1498fa0e49ab3a88973fd>:0
at OpenTK.Platform.MacOS.MonoMacGameView.RenderScene () [0x000fc] in <325fbbce19e1498fa0e49ab3a88973fd>:0
at OpenTK.Platform.MacOS.MonoMacGameView.DrawRect (CoreGraphics.CGRect dirtyRect) [0x00031] in <325fbbce19e1498fa0e49ab3a88973fd>:0
at (wrapper managed-to-native) AppKit.NSApplication.NSApplicationMain(int,string[])
at AppKit.NSApplication.Main (System.String[] args) [0x00040] in <3e1eecfdb2534f518bea4f1a21972847>:0
at MatterControlMac.MainClass.Main (System.String[] args) [0x00007] in :0
[ERROR] FATAL UNHANDLED EXCEPTION: System.TypeInitializationException: The type initializer for 'MatterHackers.MatterControl.SlicerConfiguration.ProfileManager' threw an exception. ---> System.NullReferenceException: Object reference not set to an instance of an object
at MatterHackers.MatterControl.AppContext.get_Theme () [0x00000] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController.get_Theme () [0x00000] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController..ctor () [0x00098] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController.get_Instance () [0x00007] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.PrinterInfo.set_ID (System.String value) [0x00000] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.DataStorage.ClassicDB.ClassicSqlitePrinterProfiles.ImportPrinter (MatterHackers.MatterControl.DataStorage.Printer printer, MatterHackers.MatterControl.SlicerConfiguration.ProfileManager profileData, System.String profilePath) [0x00021] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.DataStorage.ClassicDB.ClassicSqlitePrinterProfiles.ImportPrinters (MatterHackers.MatterControl.SlicerConfiguration.ProfileManager profileData, System.String profilePath) [0x00028] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager.EnsurePrintersImported () [0x00026] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager.Load (System.String userName) [0x00052] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager.ReloadActiveUser () [0x0005b] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager..cctor () [0x00016] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
--- End of inner exception stack trace ---
at (wrapper managed-to-native) System.Object.__icall_wrapper_mono_generic_class_init(intptr)
at MatterHackers.MatterControl.SlicerConfiguration.ProfileManager.add_UserChanged (System.EventHandler value) [0x00000] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController..ctor () [0x0007d] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.ApplicationController.get_Instance () [0x00007] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.DialogPage..ctor (System.String cancelButtonText, System.Boolean useOverflowBar) [0x00038] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at LicenseAgreementPage..ctor (System.Action acceptAction) [0x0000d] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.MatterControl.Application+<>c__DisplayClass5_0.b__2 (System.Object s, System.EventArgs e) [0x00023] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterHackers.Agg.UI.GuiWidget.OnLoad (System.EventArgs args) [0x0000a] in <4d176e336ae140988af12ff0893dabe2>:0
at MatterHackers.Agg.UI.GuiWidget.OnDraw (MatterHackers.Agg.Graphics2D graphics2D) [0x00008] in <4d176e336ae140988af12ff0893dabe2>:0
at MatterHackers.MatterControl.RootSystemWindow.OnDraw (MatterHackers.Agg.Graphics2D graphics2D) [0x00021] in <1762a8cd9d4c4aad9b96d09c8bb222cd>:0
at MatterControlMac.MacPlatformWindow.<.ctor>b__18_1 (System.Object s, OpenTK.FrameEventArgs e) [0x0004f] in :0
at OpenTK.Platform.MacOS.MonoMacGameView.OnRenderFrame (OpenTK.FrameEventArgs e) [0x00010] in <325fbbce19e1498fa0e49ab3a88973fd>:0
at OpenTK.Platform.MacOS.MonoMacGameView.RenderScene () [0x000fc] in <325fbbce19e1498fa0e49ab3a88973fd>:0
at OpenTK.Platform.MacOS.MonoMacGameView.DrawRect (CoreGraphics.CGRect dirtyRect) [0x00031] in <325fbbce19e1498fa0e49ab3a88973fd>:0
at (wrapper managed-to-native) AppKit.NSApplication.NSApplicationMain(int,string[])
at AppKit.NSApplication.Main (System.String[] args) [0x00040] in <3e1eecfdb2534f518bea4f1a21972847>:0
at MatterControlMac.MainClass.Main (System.String[] args) [0x00007] in :0
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

[Process completed]

Any idea how to make it work?
Thanks in advance

CoNiCiO

dmccreary, 2019-03-08:
I am having the same problem.  MatterControl will not startup on MacOS Mojave 10.14.3

conicio, 2019-03-17:
After a fresh Mojave install it works

ctkjose, 2019-04-19:
I had the exact same error, and found that you have to remove your user data from your previous MatterControl Install. 

After searching a bit I found in  MatterHackers.MatterControl.SlicerConfiguration.GetProfilesDirectoryForUser() where are these files.

There are in a ".local" folder in your home folder instead of the expected "Application Support" folder in my case:

 /Users/ctk/.local/share/MatterControl

In the terminal you can type:

mv ~/.local/share/MatterControl ~/.local/share/MatterControl_OLD

organicunity, 2022-03-23:
> I had the exact same error, and found that you have to remove your user data from your previous MatterControl Install.
>
> After searching a bit I found in [...]

This worked!  and omg, what a horrible, horrible bug.  This software crashed on boot for Windows 7, Ubuntu 20.04, and Mac OS X until this fix  I hope the devs improve this cause I want to use anything but MatterControl software, it's like turning the hobby into a torture device.

larsbrubaker, 2022-03-24:
I would love to investigate this issue more. When there is a reproduceable case I can fix most issues.

If you followed those instructions and have the MatterControl_OLD folder and can share it with me, I will try to reproduce this in a debugger.

If you can:
- Please zip the folder
- Email it or share it with me @ larsbrubaker@matterhackers.com

I understand these issues are frustrating, we work very hard to fix them and hope to address this one.

ctkjose, 2022-03-24:
[larsbrubaker](https://github.com/larsbrubaker)

Personally I haven't had this issue in a long while...

larsbrubaker, 2022-03-24:
I thought this issue was fixed. But sometime things break again. I'll leave this open for a while in case we get more info.

organicunity, 2022-03-24:
Seriously, try the install on Windows 7 or Mac OS X - heck, even Ubuntu is a nightmare to get this working.  I think it would greatly benefit your team to bundle this program with the .NET 4.7 and other dependencies and to actually show the user an error code of some sort instead of just the non-launching "show nothing" behavior.

Regardless, I did get it to work. However, V2.2 is *miles* behind V1.5, is loaded with ads, runs like molases on a i7 2017 Macbook, doesn't seem to have direct stepper motor controls, has a brutally nonintuitive awful UI and such... I uninstalled it after about 20 minutes.

Sorry, I have no interest in this program anymore and I am not interested in volunteering my time to help the company fix its broken product.

strangeluck, 2023-04-16:
I have a MacBook Pro (Retina 15-in Mid-2014) running MacOS BigSur 11.7.6. It did not have MatterControl installed previously. After install, it just hung. Force-quite and restarted a few times. All hung. I let it run hours, reading somewhere first start can be slow, but it remained hung. Reinstalls failed until I deleted the MatterControl folder here (as stated above)

~/.local/share/MatterControl 

I did a reinstall, then it started up immediately. Just an FYI.