MatterHackers/MatterControl #4293: Enable print recovery - Not acting as expected

Justin-Kirkman, 2019-02-19:
[image]

Bug Report
============
**Steps to Reproduce**
1. Make sure Enable print recovery is active
2. Interrupt a print
3. try to reprint

**Expected Behavior**
I expected when I opened MatterControl it would notify me of the interrupted print and I could reprint.

**Actual Behavior**
When I opened MC, there was no message until I connected to the printer.  After I connected and the message appeared, I expected it to print from where I left off. (which appears what it was trying) When it tried to reprint it started hitting the edge of the previous printed piece and I had to stop the print. 

**MatterControl Build Number**
Version 2.19.1.10116

**Operating System Version**
Windows 10 pro
**Printer Make/Model**
Pulse Xe

larsbrubaker, 2019-02-20:
Would be good to check the printer on open and tell there is a print recovery available even without connecting.

martinpirringer, 2019-05-03:
There is a workaround I sent to support a while back. Before you tell the printer to resume almost execute the Startup Gcode or manually if you have an LCD autohome and make sure the temps are correct. You can set the resume code temporarely to something like this (depending on your printer you might have to make changes

G28 ; home all axes

;M200 D1.75 ; Volumetric set to 1.75mm filament - don't use unless you have a slicer that does volumetric like SLIC3R

M104 S[temperature] ; Turn Nozzle on

M190 S[bed_temperature] ; Turn Bed on and wait

M109 S[temperature] ; Turn Nozzle on and wait
; The M104, 190, 109 is to save time to heat both bed and nozzle simultaneously. It is neccesary to turn the nozzle an bed back on as otherwise the printer might have (or will have if it lost power) the nozzle turned off and resuming the code does not turn it back on as you just continue with the Gcode
G92 E0; Reset extruder. 
G90; Turn absolute addressing on

Drawback of this workaround is that now every time you click stop and then resume the printer auto homes. So there should be a macro to "recover" from errors and power outage etc. Same procedure can help you recover from a thermal runaway caused by lets say the fan blowing to hard on the nozzle or bed like it does on the pulse