MatterHackers/MatterControl #4319: Printer finishes before Matter Control

blueeagle69, 2019-03-01:
Bug Report
============
**Steps to Reproduce**
1.  Print via USB
2. 
3. 

**Expected Behavior**
When printer has finished printing object, MC should show completed.
**Actual Behavior**
Matter Control shows 96% completed and cannot cancel print job.
Object has printed successfully. 
**MatterControl Build Number**
2.192.2.10169

**Operating System Version**
Windows 10 Pro
**Printer Make/Model**
Anet A8 (stock board) with Marlin 1.1.9
[image]

blueeagle69, 2019-04-27:
Hi.

Apologies.
After a brief hiatus while I upgraded my Anet A8 to a bowden system, I took a look at my end Gcode, and I noticed there was a G4 wait command before the heaters were disabled.
I removed  it, and it has appeared to have fixed the issue,
My bad, I didn't notice it until now.

closed 2019-04-27 by blueeagle69

larsbrubaker, 2019-04-27:
Glad to hear it. Thanks