MatterHackers/MatterControl #4402: Debug MatterSliceLib

NickNaNick, 2019-04-05:
Hello

I would really like to understand what is going on in the slicing process. So I need to debug the MatterSliceLib. I have the problem, that my breakpoints are not hit, for example in the fffProcessor.

I cloned MatterControl, then cloned agg-sharp and MatterSlice and put them into the Submodules folder. 

Am I missing something?

Thanks for your help
Best regards,
Manuel Aeschbacher

larsbrubaker, 2019-04-05:
If you add the file: MatterControl.json to your user folder (mine is C:\Users\larsb) with the following content:

```
{
  "MatterControl": {
    "Slicer": {
      "Debug": true
    }
  }
}
```

You will be able to debug MatterSlice the way you expect. MatterSlice runs as a separate program due to its licensing, and we are very careful to make sure how we handle dependencies around it. 

Hope this helps.

NickNaNick, 2019-04-09:
Thank you, now it works. I noticed also, that I had to load a new STL-File, with the one which was already loaded into the start screen it did not hit the breakpoints.
It is also possible to debug with rightklick on a NUnit testfile in the testexplorer and click "debug selected test". That would be sufficient to explore the functionality of a certain module.

Have a nice day and thanks for the software :)

closed 2019-04-09 by NickNaNick