MatterHackers/MatterControl #4496: Align to original modeled coordinates

teejaydub, 2019-05-03:
Feature Request
===============

**Problem**
Setting up a dual-extruder print: You can import multiple STLs and indicate which one is which material, but there's no way to get the parts aligned the way they were modeled.

**Solution**
Restore the old "Align" feature, which returns the selected models to their original STL file coordinates. It's been replaced in 2.0 by something different with the same name, so you could call this "Align to original coordinates" or "Align to model origin" or "Align for multiple extrusion."

larsbrubaker, 2019-05-03:
This is still in there. Here is a gif of aligning to mesh coordinates:

[image]

teejaydub, 2019-05-03:
Ah, got it, thanks. 

I did see those Origin buttons (after I made the palette wider; they were hidden originally) but I was distracted trying to do something more complex - import one part twice, rotate one of the duplicates, then align them at their original origins. I'll just do the rotation as a separate step in my modeling program.

This will work fine for the general case.

Thanks!

closed 2019-05-03 by teejaydub

mstange42, 2019-05-15:
Thank you very much, this is something I had not found!