MatterHackers/MatterControl #4501: Sync'ing Toggle

mstange42, 2019-05-04:
Feature Request
===============

**Problem**
Syncing to multiple computers makes it difficult to do work. 
Ex: Failed print on 1 machine. Have made changes while testing another model. Forces a reslice on the print machine, and if change were not made to restore it has wrong settings. 
The current solution of logging off does not really work for me, as when above changes are what I want, I want the setting to be synced, but I cant without logging in and redoing the whole settings. 

**Solution**
Instead of forcing a sync when anything is changed on any computer have a toggle to activate/deactivate the syncing.

jlewin, 2019-05-06:
Are you trying to use a single profile for multiple printers? Currently we expected dedicated profiles for each printer and as such you should never have two MatterControl instances connected to a given printer at the same time

mstange42, 2019-05-06:
No, I only have 1 printer.