MatterHackers/MatterControl #4530: Add ability to run nozzle calibration and xy calibration without a z-probe

larsbrubaker, 2019-05-15:
These should still work and will add value to dual extrusion printers. At the moment they only show up if there is a z-probe installed.

jlewin, 2019-05-15:
Do we imagine that on a machine without a probe the wizard would still be called the `Probe Calibration Wizard`? My frustration with the current arrangement is due to this exact issue. In my mind both of these wizards perform `Z Calibration` and `XY Calibration`, where the user manually has to observe some real world results and supply them back to MatterControl.

larsbrubaker, 2019-05-15:
Your thinking seems correct to me. I feel they should be called Z Calibration and XY Calibration, and apply in non-probe situations as appropriate. 

Further, if we get the fully auto versions working with a probe they would still just be called the same thing.