MatterHackers/MatterControl #4538: Nozzle Wipe

blueeagle69, 2019-05-16:
Feature Request
===============
Nozzle Wipe after Retract
**Problem**
According to the online manual, the option should be there.
Try as I might I cannot find it.

**Solution**
Explain how this feature will solve the problem.
N?A

unlimitedbacon, 2019-05-17:
This setting was removed because it was not useful.

I presume you found this through the [old MatterControl Wiki](http://wiki.mattercontrol.com/SETTINGS/Filament/Retraction/While_Printing/Wipe_After_Retract). I added a note to the page for that setting marking it as outdated. You can find the [new MatterControl documentation here](https://www.matterhackers.com/mattercontrol/support), although the page on retraction settings has not been written yet.

blueeagle69, 2019-05-17:
Cheers, I clearly missed it!

closed 2019-05-17 by larsbrubaker