MatterHackers/MatterControl #4574: Started using my printer at home and it reverted to an old web setting that lost all my recent changes

larsbrubaker, 2019-05-25:
So much tuning just disappeared. We have to fix this. Terrible experience.

jlewin, 2019-05-28:
This should be very reproducible 
- Be on debug at latest profile version
- Run release, should update and be synced with debug
- Make changes in Release, release should be latest server version
- Run debug - debug should push old settings as latest, reverting any recent changes in release

MPL1963, 2019-05-30:
Same thing has been happening to me the last two weeks. Steps to repro:
1. Add a new Material.
2. Change print settings.
3. Print file, then close app.
4. Open app. New Material and changed settings are gone.

larsbrubaker, 2019-05-30:
This has been fixed in the latest Alpha build and will be in the next release.

closed 2019-06-04 by larsbrubaker