MatterHackers/MatterControl #4579: Subtract not working (even with two cubes), works with

schorhr, 2019-05-27:
Bug Report
============
**Steps to Reproduce**
1. Create two overlapping cubes
2. Select both cubes and click on subtract
3. Choose which geometry to subtract 

**Expected Behavior**

Cube cuts cube

**Actual Behavior**

Cube does not cut cube's geometry

**MatterControl Build Number**

MatterControl versions:
2.19.2.10169 Does not work.
2.19.2.10116 Does not work
2.0.0.9794 = Works, but crashes very frequently 😉

**Operating System Version**

Windows 10 Pro 64 Bit Build 17134.706
Windows 10 Pro 64 Bit Build 17134.648

Hardware: Intel E8500=3.16GHz/4-6GB/160GB

10 out of 12 computers show this issue (A PC where it's working has .648, but one also 706)
. They are all the same, minor differences might be versions of .net, Java and such. Graphic drivers are all the same version.

**Printer Make/Model**
(Does not apply; Ender 3, Anet A8)

*Thank you for your hard work. MatterControl is an amazing program, and it's great for use in a school workshop!*

MDAR, 2019-10-07:
I can second this.

I managed to create a shallow cube with 4 holes in a previous version, but since an update, I can't do this.

The subtract button only creates a new group of the two items, that can be "ungrouped"

neil969, 2020-02-03:
I am finding this same problem now since the  update to V2.20.1.10422 
It just is now IMPOSSIBLE to subtract anything from anything.. It just groups them and the "subtract  job ( with a little egg timer) sit at the bottom forever .. :( 
This is REALLY sad becasue this makes the application completely useless and I have been using daily for more than a year -
The image shows the result of attempting to subtract a small arc from this object ( imported as an STL)
 Windows10  
[image]

MDAR, 2020-02-04:
My apologies.

I forgot to update the thread when I discovered the missing step.

Instead of my creating a video, just watch the first minute or so of this one.

https://youtu.be/f_MHta2Tivc

Basically, a fabulous option box appears on the right that we need to interact with now, that has lots of options.

Good luck

neil969, 2020-02-04:
Thanks. I have been using the application for quite a while and the tools are all good, and I have done many. many subtractions ;) 
The Issue I see here seems to be related to the complexity of the  object from which I am trying to subtract.  Matter Control had , in earlier editing created some errors in the object, that were "unfixable"  ( Slicers just made things disappear when slicing LOL!)
When attempting to subtract from the "complex" object, the subtract routine just seems to crash silently.
I started the whole job again, working in only small bits and saving often, and again the object ended up with errors ( one so bad that the whole application crashed when I tried to even LOOK at the library of object that I saved - I had to rename the object to get the program to stop crashing! - I  will attach the file - in the zip , rename the file by taking the "OLD_"  out of the extension . 

[Pin-hole_cover_2.zip](https://github.com/MatterHackers/MatterControl/files/4154483/Pin-hole_cover_2.zip)

CommanderRedYT, 2020-09-26:
Have kinda the same issue, I'm on Ubuntu 20.04.1 (AMD Ryzen 5 3500U, processor graphics) and I can only subtract built-in objects like the cube and cylinder... When I import an .stl it just merges it / does nothing

arachaic, 2021-01-05:
Having the same issue even after making it a component. I get this error.

[image]

larsbrubaker, 2022-04-20:
This has been fixed

closed 2022-04-20 by larsbrubaker